FormerLurker/Octolapse #844: Default Directory Error for Timelapse Folder (Was working and always set to Default)

JOBNED1, 2022-04-26:
#### If this is a feature request describe it here
This is not a feature request. This is an issue that was working previously but somehow gained a fault and I'm not seeing a way to resolve it because I've tried what I could find. including but not limited to uninstalling and reinstalling octolapse.

#### Version of Octolapse
Octolapse Version: 0.4.1

#### Version of OctoPrint
OctoPrint Version: 1.7.3

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred
1 - I was starting another print and the snapshot didn't appear.
2 - I have both started lapse prints from my octopi(octoprint) and from cura thru the octoprint plugin successfully but this time it failed to even try to start.

#### What should have happened?
It should have gave me the snapshot screen on the octoprint page and then started the print like it has several times before.

#### What happened instead?
It gave an error and froze the print no matter what source I tried to start the time lapse print from and gave me the message to test my directories (which are set to default by the way). Again it was working successfully prior to this but no g code has successfully started a lapse or give a different result since this. I am able to print normally with octolapse disabled however.

#### Operating System running OctoPrint and Octolapse
OS Name: Octopi
Os Version: Version 0.18.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

#### Printer model & used firmware incl. version
Printer Model: Voxelab Aquilla
Printer Firmware Version: https://github.com/alexqzd/Marlin/releases UBL-BLTouch-10x10-G32.bin

#### Browser and version of browser, operating system running browser
Browser: Firefox or Chrome Browser or Octoeverywhere (Don't think this is the problem but don't know)
Browser OS: Windows 10 or Windows 11 or Android phone

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: As I said since the first G code. Every G code I tried failed even ones I've successfully got time lapses of before. but I will send this if you think it is required after the initial response.

#### Link to settings.json
Link to settings.json with all passwords removed: https://pastebin.com/W8ZCjHfS (Camera ha no login [pi cam])

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://pastebin.com/S3VuRgr7

#### Link to octoprint.log
Link to octoprint.log:   [File to large to add to pastebin...]

#### Link to contents of Javascript console in the browser
Link to javascript console output: (You may have to help me to this one but I do know that I can't even mkdir in PuTTY without it saying it failed...)

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: Haven't took any but will if they end up being needed later.

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

JOBNED1, 2022-04-28:
Went in and edited the pastebin as unlisted. It should be accessible now. Sorry about that. Though that is the great thing about having an account even if it is a free account. If you are logged in you can go back and edit them and change things without having to redo the whole thing link and all.

JOBNED1, 2022-04-28:
This is what I get in response testing the finale two folders. And yes I know it is currently disabled I did that so I could print files until whatever happened is fixed.
[image]

This is what I get since it stopped printing for testing the first folder.
[image]

I also get this message and I have created that directory in the pi via PuTTY (I finally got mkdir to work on their.). So I know that that folder is actually already there so it shouldn't have to make the directory.
[image]

FormerLurker, 2022-04-28:
Looks like something is messed up with the folder permissions, but there isn't enough logging there to tell exactly what's going on.  However, if you have sftp (can also be done with putty), check the /home/pi/.octoprint/data/octolapse folder permissions.  They should look like this:

[image]

You'll need to apply the permissions recursively.

Also, if you find a directory in there called `octolapse_test_temp`, delete it.

Then look at the default timelapse folder:

[image]

Verify it also has the same permissions applying whatever is missing recursively, and delete `octolapse_test_temp` if you find it.

Not sure what happened here, but Octolapse doesn't change any permissions.  I do need to improve the error reporting/logging when testing for permissions, but this is the first time I've had any issue reports about this functionality.

JOBNED1, 2022-04-28:
Here is what I'm seeing in the case of what permissions Octolapse has in folder form...
[image]

and here is the permissions inside Octolapse itself...
[image]

JOBNED1, 2022-04-28:
Okay checked the Timelapse folder's permissions.
[image]

I also found the one you told me to delete so I removed it.
[image]

JOBNED1, 2022-04-28:
Okay so with that removed (which is the only thing I did). It seems I can now test the folders successfully in the settings and saved the settings I'm going to do a test print with octolapse soon. I will comment again and close if that successfully fixed the error. If the error happens again I'll let you know in the next comment and we can take another look for causes.

FormerLurker, 2022-04-29:
You have exposed a flaw in my test that I can now correct!  I need to attempt to delete the test folder.  Looks like that failed for whatever reason for your setup, which will cause it to forever fail thereafter.  Will fix asap and post back then.

JOBNED1, 2022-04-29:
Well it seems that worked. The first print I did made a Timelapse without issue again and it is saved. Thank you for the help and I'm glad I was able to help you make the plugin better as well. If anything else happens that seems to fowl up Octolapse I'll return.

closed 2022-04-29 by JOBNED1