FormerLurker/Octolapse #848: [Request] firmware retract using G10 G11

mikeoverbay, 2022-06-14:
My printers (3) are all setup to use firmware retraction. To retract, sending a G10 is all that is needed. G11 recover it is g11.
No matter how I set up the values in retraction in Octolapse, my printers are ignoring them. Probably because FW retraction is enabled on my machines.

A simple "Use FW Retract" button and sending G10 before the move to snapshot and a G11 after returning would be all that is needed.
Hops are controlled in the M208/M207 commands. It is the only way to adjust them with FW retraction enabled in Marlin other than at the machines terminal.

geekazine, 2022-08-10:
Listening. I also want a good pre/post script to not only do G10/G11 retraction, but allow me to slow down or pause the print head before moving. I get way too much stringing when I use OL.

edgauthier, 2022-10-03:
+1 for firmware retraction support.