FormerLurker/Octolapse #850: module 'PIL.Image' has no attribute 'open'

funkypenguin, 2022-06-20:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 0.8.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Run a print in debug mode

#### What should have happened?

Image snapshots should have been created

#### What happened instead?

Errors were recorded in the log:

```
2022-06-20 07:49:27,482 - octolapse.snapshot - ERROR - An unexpected exception occurred while creating a snapshot thumbnail for the Webcam - Default OctoPi 0.16.0 camera.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/snapshot.py", line 470, in create_thumbnail
    with Image.open(latest_snapshot_path) as img:
AttributeError: module 'PIL.Image' has no attribute 'open'
2022-06-20 07:49:27,498 - octolapse.snapshot - ERROR - An unexpected exception occurred while post-processing images for the Webcam - Default OctoPi 0.16.0 camera.  See plugin.octolapse.log for details
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/snapshot.py", line 470, in create_thumbnail
    with Image.open(latest_snapshot_path) as img:
AttributeError: module 'PIL.Image' has no attribute 'open'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/snapshot.py", line 339, in process
    self.create_thumbnail()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/snapshot.py", line 490, in create_thumbnail
    cause=e
octoprint_octolapse.snapshot.SnapshotError: snapshot-thumbnail-create-error: Create Thumbnail - An unexpected exception occurred while creating a snapshot thumbnail for the Webcam - Default OctoPi 0.16.0 camera.  Check the log file (plugin_octolapse.log) for details. - Inner Exception: module 'PIL.Image' has no attribute 'open'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/snapshot.py", line 545, in post_process
    image_post_processor.process()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/snapshot.py", line 349, in process
    self.on_post_processing_error_callback(e)
TypeError: on_post_processing_error_callback() missing 1 required positional argument: 'exc'
```

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: N/A
Printer Firmware Version: N/A

#### Browser and version of browser, operating system running browser

Browser: N/A
Browser OS: N/A

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: N/A

#### Link to settings.json

Link to settings.json with all passwords removed: I wasn't able to find how to export `settings.json`

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/9vLy1YJs

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/yNMtuXng

#### Link to contents of Javascript console in the browser

Link to javascript console output: N/A

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: N/A

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

geoscoutCJ, 2023-07-21:
I am running into this problem as well. No snapshots are being saved correctly. I am usually able to complete the video. I am also having issues with the stabilization not working fully but I think that is a different issue that I just need to work on my settings.

_Version of Octolapse:_
Version: 0.4.3

_Version of OctoPrint_
OctoPrint Version: 1.9.2

_When you ran into the problem, did you have diagnostic logging enabled?_
Yes.

_What were you doing when the problem occurred?_
Run a print in test mode.

_What should have happened?_
Image snapshots should have been created

_What happened instead?_
Errors were recorded in the log and no images were captured.

_Operating System running OctoPrint and Octolapse_
OS Name: Octopi
Os Version: 1.0.0

_Printer model & used firmware incl. version_
Printer Model: Prusa MK3S
Printer Firmware Version: 3.12.2-5713

_The gcode file you were printing when the problem occurred._
Attached in ZIP.

_Log file plugin_octolapse.log._
Attached in ZIP.

[PIL.Image_Issue.zip](https://github.com/FormerLurker/Octolapse/files/12132510/PIL.Image_Issue.zip)

geoscoutCJ, 2023-07-26:
I was able to fix my image stabilization and can create a time-lapse, but I still do not have any preview snapshots.