FormerLurker/Octolapse #856: Octolapse only searches 1000 gcode lines, which fails with thumbnails

ondrejch, 2022-07-22:
When the thumbnail gcode is over 1000 lines, octolapse cannot detect the automatic settings provided by Cura. Maybe it should skip thumbnails before looking for the settings in gcode?

BalooDK, 2022-08-05:
I decreased my thumbnail size for this same reason
You can also put the script in your end gcode

ondrejch, 2022-08-05:
I just patched the code, so it scans the whole GCODE file.

TrainAss, 2022-09-11:
@ondrejch how did you patch the code?

ondrejch, 2022-09-11:
Just increased the line limit.

TrainAss, 2022-09-13:
I'm not sure how to do that, sorry.

ondrejch, 2022-09-13:
Change the number in settings.py, lines with 
search_direction="both", max_forward_search=100000, max_reverse_search=1000

TrainAss, 2022-09-14:
Awesome, thank you very much!

vaderag, 2023-02-11:
I presume this is the same issue I just saw - I only just enabled thumbnails and the moment I tried to print I got an error...