FormerLurker/Octolapse #863: Cannot start Octolapse with print

TrainAss, 2022-09-03:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1+u.d33011e

#### Version of OctoPrint

OctoPrint Version: 1.8.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Octolapse was enabled
2. Selected job stored on internal storage to print 
3. Clicked print.

#### What should have happened?
Octolapse should have analyzed the GCode and presented the snapshot plan window and then the print starts.

#### What happened instead?
Received an error and the print does not start

#### Operating System running OctoPrint and Octolapse

OS Name: Ubuntu Server LTS
Os Version: 22.04.1

#### Printer model & used firmware incl. version

Printer Model: Ender 3 v2
Printer Firmware Version: mriscoc professional firmware Ender3V2-422-BLTUBL-20220816

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [GCode File](https://pastebin.com/NTHkiJqV)

#### Link to settings.json

Link to settings.json with all passwords removed: [settings.json](https://pastebin.com/wgwZ84rS)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [Octolapse Log](https://pastebin.com/shc0KaXU)

#### Link to octoprint.log

Link to octoprint.log:  [Octoprint Log](https://pastebin.com/HPUmwWzT)E

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Ruhel786, 2023-03-18:
**Version of Octolapse**
Octolapse 0.4.1  Settings Version: 0.4.0

**Version of OctoPrint**
Version 1.8.7

**When you ran into the problem, did you have diagnostic logging enabled?**
Diagnostic Logging was Enabled: YES

**What were you doing when the problem occurred**
Octolapse was enabled
Selected job stored on Raspberry Pi 4 internal memory
Clicked print

**What should have happened?**
Octolapse should have analyzed the GCode and presented the snapshot plan window and then the print starts.

**What happened instead?**
Received an error and the print does not start

**Operating System running OctoPrint and Octolapse**
OctoPi v1.0.0-1.8.6-20230301124943

**Printer model & used firmware incl. version**
**Printer Model:** Prusa i3 MK3S
**Printer Firmware Version:** v3.11.0

**Link to the gcode file you were printing when the problem occurred
Link to Gcode File:** [PLA_2h53m_upper_case.gcode](https://gist.githubusercontent.com/Ruhel786/8b26a790ac9abc55c198a175aff88620/raw/4dfeb2c68deaff9140c73cd56edc8df980a28844/PLA_2h53m_upper_case.gcode)

**Link to settings.json
Link to settings.json with all passwords removed:** [settings.json](https://gist.githubusercontent.com/Ruhel786/8b26a790ac9abc55c198a175aff88620/raw/4dfeb2c68deaff9140c73cd56edc8df980a28844/settings.json)

**Link to plugin_octolapse.log
Link to plugin_octolapse.log:** [Plugin_Octolapse Log](https://gist.githubusercontent.com/Ruhel786/8b26a790ac9abc55c198a175aff88620/raw/4dfeb2c68deaff9140c73cd56edc8df980a28844/plugin_octolapse.log)

**Link to octoprint.log
Link to octoprint.log:** [Octoprint Log](https://gist.githubusercontent.com/Ruhel786/8b26a790ac9abc55c198a175aff88620/raw/4dfeb2c68deaff9140c73cd56edc8df980a28844/octoprint.log)

Screenshots and/or videos of the problem:
Screenshot/Video Links: None taken as yet

FormerLurker, 2023-04-02:
@Ruhel786, your issue looks like your camera profile isn't able to get a snapshot.  Please edit your camera profile, and click the 'test' button next to the snapshot url, and adjust as needed.

@TrainAss can you try disabling smartpreheat and commenting out these lines from your gcode:

M190 S{{ (plugins.smartpreheat.bed * 0.8)|round }}
M140 S{{ plugins.smartpreheat.bed }}
M104 T{{ tool }} S{{ temp }}
M190 S{{ plugins.smartpreheat.bed }}
M109 T{{ tool }} S{{ temp }}

If that works, you can add them back and try adjusting this section of your octolapse printer profile to include the appropriate number of extruders:

[image]

Ruhel786, 2023-04-02:
@FormerLurker Hi. This seems to be an intermittent issue as if I reset my PI with OctoPrint on it the issue will resolve until a few days later. A quick reset will fix the issue. When the issue does occur I can still see the camera in the preview window and the snapshot test still works.

FormerLurker, 2023-04-02:
Thanks, I will circle back regarding this asap.