FormerLurker/Octolapse #864: Octolapse and PrusaSlicer 2.5 Jamming

jimmysdiy, 2022-09-10:
I have been using OctoLapse for over a year with Cura. I recently switched to PrusaSlicer and setup OctoLapse according to the wiki. The printer (ender 3 w/ Micrswiss DD Kit) stops extruding after 1st layer due to retraction which pulls the filament too back I feel causing it to jam.

Versions
PrusaSlicer 2.5 w/ Verbose enabled
OctoPrint Latest w/ G90/G91 enabled under Features because I use your Arc Welder Plugin
Marlin 2.1.1 firmware
Octolapse - Latest version

I have tried automatic configuration w/ verbose
I have tried the manual setting for Prusa by copying all info 
retraction - 0.5mm
retraction/de-retraction speed 35mm
z lift - 0.2mm
Speed - 100mm based on my prusa

no change. If I do custom profile enable, then making G90/G91 enable in Octolapse printer setting (nothing prints).

jimmysdiy, 2022-09-12:
Ok, I was digging into the Ender 3 Profile. Seems like it turns retraction to 9000 for the Slic3r PE/Prusa Slicer Rest stays the same. This results in retraction being too much. Going to mod it and upload it into OctoPrint and see if that fixes it

jimmysdiy, 2022-09-16:
Ok, since i got Marlin2.0+, I had to enable "G90/G91 overrides relative extruder mode Smoothieware Marlin 2+
This is used by the GCODE analysis in the backend and the viewer in the frontend to interpret your sliced files correctly" in OctoPrint. Octolapse doesn't read this setting from OctoPrint and it continues to go w/ False causing these jams.

Fix - I downloaded the profile, edited with Notepad++ to and changed to G90/G91 - True

worked!

Darkstar2k2003, 2022-11-15:
I am having this exact issue as well.

Creality CR-10s with skr mini e3 v3 main board
Micro Swiss NG Extruder and Hotend
PrusaSlicer 2.5 w/ Verbose enabled
OctoPrint v.1.8.6 
Arc Welder Plugin
Marlin 2.1.1 firmware
Octolapse - 0.41