FormerLurker/Octolapse #872: I can't install Octolapse, it keeps crashing.

xarayy, 2022-11-21:
hello I can't install Octolapse, I'm getting this error. I do not use raspberry, I am trying to install it on android tvbox, could the problem be caused by this? I installed dozens of plugins, I just had problems with octoplapse

[image]

FormerLurker, 2022-11-21:
Guessing you are running Octo4a? In the app, in the settings menu, there is an option called "install plugin extras". Try that option and let me know if it helps.

crazydomi, 2023-02-03:
I have the same problem, but installing "plugin extras" did not help. I installed Octo4a with https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359 .
Other plugins like OctoEverywhere or BedVisualizer are running.

sv0o, 2023-02-23:
I've been trying to install octolapse on a jide remix (android tv box) and running into similar problems.  
A possible fix was proposed here but I couldn't get it to work:
https://github.com/feelfreelinux/octo4a/discussions/370

sv0o, 2023-04-03:
I just tried up update from the a-c-m fork to 0.2.4

here's what the plugin manager console said:
`                                                                                                                                                                                                                                                                            2023-04-03 16:39:00,486   /usr/bin/python3 -m pip --disable-pip-version-check install https://github.com/FormerLurker/Octolapse/archive/v0.4.2.zip --no-cache-dir
2023-04-03 16:39:09,129 > Collecting https://github.com/FormerLurker/Octolapse/archive/v0.4.2.zip
2023-04-03 16:39:10,545 > Downloading https://github.com/FormerLurker/Octolapse/archive/v0.4.2.zip (1.3 MB)
2023-04-03 16:39:12,029 > ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.3/1.3 MB 1.0 MB/s eta 0:00:00
2023-04-03 16:39:14,506 > Preparing metadata (setup.py): started
2023-04-03 16:39:27,671 ! error: subprocess-exited-with-error
2023-04-03 16:39:27,676 ! 
2023-04-03 16:39:27,679 ! × python setup.py egg_info did not run successfully.
2023-04-03 16:39:27,683 ! │ exit code: 1
2023-04-03 16:39:27,686 ! ╰─> [10 lines of output]
2023-04-03 16:39:27,689 ! Argon2 passlib backend is not available, not using it for password hashing
2023-04-03 16:39:27,692 ! /tmp/pip-req-build-uv1vaa44/setup.py:58: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
2023-04-03 16:39:27,695 ! if LooseVersion(octoprint.server.VERSION) < LooseVersion("1.4"):
2023-04-03 16:39:27,698 ! error in Octolapse setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after version specifier)
2023-04-03 16:39:27,701 ! pillow >=6.2.0<7.0.0
2023-04-03 16:39:27,704 ! ~~~~~~~^
2023-04-03 16:39:27,707 ! likely tags: v0.4.2
2023-04-03 16:39:27,710 ! picking v0.4.2
2023-04-03 16:39:27,713 ! got version from expanded keyword {'version': 'v0.4.2', 'full-revisionid': 'c6c86782cdbfc3590fa4b1bf2e63064eb459b16b', 'dirty': False, 'error': None, 'date': '2023-04-02T10:23:21-0500'}
2023-04-03 16:39:27,716 ! Found packages: {'octoprint_octolapse', 'octoprint_octolapse.test', 'octoprint_octolapse_setuptools'}
2023-04-03 16:39:27,719 ! [end of output]
2023-04-03 16:39:27,722 ! 
2023-04-03 16:39:27,725 ! note: This error originates from a subprocess, and is likely not a problem with pip.
2023-04-03 16:39:27,728 ! error: metadata-generation-failed
2023-04-03 16:39:27,731 ! 
2023-04-03 16:39:27,734 ! × Encountered error while generating package metadata.
2023-04-03 16:39:27,737 ! ╰─> See above for output.
2023-04-03 16:39:27,740 ! 
2023-04-03 16:39:27,743 ! note: This is an issue with the package mentioned above, not pip.
2023-04-03 16:39:27,746 ! hint: See above for details.
2023-04-03 16:39:28,256 > Preparing metadata (setup.py): finished with status 'error'
`