FormerLurker/Octolapse #882: Render Error

obihann, 2023-02-18:
#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version:  1.8.6

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Trying to render a print

#### What should have happened?
Print should have rendered

#### What happened instead?
```
Rendering failed for camera 'Webcam - Default OctoPi 0.16.0'. render-error: Unknown render error. Please check plugin_octolapse.log for more details.. Inner Exception: not enough arguments for format string
```

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian GNU/Linux
Os Version:  10 (buster) armv7l

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/obihann/806ad01225ce5b5dfd03aea5b2c9bcec

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [LINK_GOES_HERE](https://gist.github.com/obihann/7bf1c9794fda6a3e373261ffc7a25372)

#### Link to octoprint.log

Link to octoprint.log:  [___REPLACE_THIS__LINK_GOES_HERE](https://gist.github.com/obihann/b4641d398cd8a88160bc7f6a619dd436)

obihann, 2023-02-18:
[settings.json](https://gist.github.com/obihann/3a2e0246190a5dc07ff04b05738bd982)

Duetting, 2023-02-24:
I get the same error.
It seems to come from a percent sign (%) inside the print job name (Cura GCode File Name Format default adding the Infill percentage).
The used string interpolation with RenderJobInfo.snapshot_filename_format misses to double up (escape) the percent sign.
If i find the time i will try to fix it and create a pull request, but Iäm not a python expert and this variable is used in a few places and probably pre_roll_snapshot_filename_format has the same problem. 
The easiest workaround is very likely to not have percent signs inside print job/gcode filenames.

Duetting, 2023-02-24:
Someone already fixed it, the pull request was just not merged up to now:
#852