FormerLurker/Octolapse #883: Constantly asking me for permission to update profile

SHoover80, 2023-02-21:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: (0.4.1+u.215845c)

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.8.6

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. Anytime I log in to Octoprint
2. Anytime I refresh the page
3. Restart Octoprint

Anytime I do these actions I get notifications for Octolapse in Octoprint asking if I want to apply updates. I have made no recent changes to my Octolapse profiles. 

[image]

I have to keep choosing OK and apply the profile. I get no errors after applying the update or indication of any errors other than constantly seeing this annoying notification. 

[image]

When I choose Ignore I do get this error:

[image]

#### What should have happened?
Not seeing this notification constantly. 

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: Windows 11
Os Version: Version 22H2 (OS Build 22621.1265)

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Version 110.0.5481.77 (Official Build) (64-bit)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Stonemercy, 2023-03-13:
Encountering this too, hope there's a fix soon

Pseudo-Nym, 2023-03-23:
I have this happening on one instance of Octoprint. There are two profiles setup for Octolapse one is for Creality - Ender 3 Pro the other, Creality CR-10 V3.

SHoover80, 2023-03-23:
Yea, that's what's got me, I only have one profile setup for one printer.

FormerLurker, 2023-04-02:
Can someone export their settings file and upload it (preferably to gist.github.com):

[image]

I've checked the settings repo, and can't find an issue, but obviously something is going on.

Stonemercy, 2023-04-02:
[settings.txt](https://github.com/FormerLurker/Octolapse/files/11133074/settings.txt)
Changed from json to txt cos github formatting

FormerLurker, 2023-04-02:
@Stonemercy, I have a theory.  Thanks for the settings!

Try this:

Edit your "Smart - Snap To Print - High Quality" trigger profile.  un-check 'Customize Profile' and accept that the settings will be overwritten.  Save it.  Now see if you get that message.

I'm working to verify that, but I think maybe the update check is happening with custom profiles, even though it should not be.  It doesn't seem to actually overwrite them, but it might be reporting on newer versions anyway.

FormerLurker, 2023-04-02:
Here is a workaround for everyone:

[image]

Then uncheck this box to prevent profile updates:

[image]

You can re-enable them after I've figured this out and fixed it.

Stonemercy, 2023-04-02:
@FormerLurker I'm currently using the Smart - High Quality profile so idek why Smart - Snap To Print - High Quality is customised 😆

I've unchecked the customization and rebooted (and I noticed it needed to be updated it to 4.2)

After the two updates and turning back on Automatic Updates (I had done the workaround about a week ago) it seems to have worked 👍

FormerLurker, 2023-04-02:
@Stonemercy,

Thanks for confirming my guess!  I actually haven't been able to trigger that issue locally, but it was the only profile that was behind, so it made sense.  I'll fiddle with it, but be sure to let me know if you get another annoying notification that doesn't go away after you update!  I will be adding a bunch of printer settings soon, so it's possible a legit update will be pushed soon, just FYI.

Stonemercy, 2023-04-03:
[image]
Unfortunately this popped up this morning 😢