FormerLurker/Octolapse #886: Video stops before end of print

fractionalJoe, 2023-03-10:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.8.7

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Installed Octolapse
2. Configured for Prusa MK3S+ using Animated Printing, Smart - High Quality, MP4 - 60FPS, 
3. Printed https://www.printables.com/model/229085-medieval-castle

#### What should have happened?
It should have created a timelapse from beginning to end

#### What happened instead?
There was a long pause at the beginning of the timelapse (no movement) and the video stops at the end before the print is finished. (The print finished successfully without any pauses or anything like that.)

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi / Raspbian
Os Version: OctoPi 1.0.0

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3S+
Printer Firmware Version: 3.12.2

#### Browser and version of browser, operating system running browser

Browser: Edge
Browser OS: Windows

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://1drv.ms/u/s!ApdCL-1qpFItgaV96I9QgCAEpwjEIg?e=DyxHpl

#### Link to settings.json

Link to settings.json with all passwords removed: https://1drv.ms/u/s!ApdCL-1qpFItgaV-66aKHL8ncMHVbA?e=uPLEJX

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://1drv.ms/u/s!ApdCL-1qpFItgaV_DJdRxggIdgNhZw?e=QBcdp3

#### Link to octoprint.log

Link to octoprint.log:  https://1drv.ms/u/s!ApdCL-1qpFItgaYA5c0juOZqmt6LLw?e=zOMg6a

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://1drv.ms/v/s!ApdCL-1qpFItgaV8YyqA7kXJVha53A?e=PJwNsR

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2023-04-02:
The smart triggers currently don't take an image of the empty bed.  I am working on that.

The 'pause' at the beginning and end is controlled in the 'rendering' profile as 'pre' and 'post' roll.  You can turn that off it you want.  I added that because people reported that as a bug (starts and stops too suddenly).  I guess you can't please everyone, lol!

The last layer has an image, but the layer isn't complete due to the way the smart triggers work.  They GREATLY improve quality, but come at the cost of the last frame.  I am actually working to fix this problem now, but it's trickier than one might think.  It will be available in the next release.

Anyway, thanks for reporting this, and sorry for the delay.  I'm playing catch-up right now.

fractionalJoe, 2023-04-06:
Thanks @FormerLurker!

ninjamojo, 2023-04-06:
Checkout Issue #889, which contains info for how to install a dev feature where you can insert ```@OCTOLAPSE TAKE-SNAPSHOT``` wherever you what in your gcode even while using the smart triggers (layer, snap to print, etc...).  It's not 100% automatic, but it's pretty easy.  I added mine right before the priming line goes down, but you can put it wherever.  Then I added another in my end gcode.

See if this works for you.