FormerLurker/Octolapse #887: PrusaSlicer 2.5.1 and stop printing after 1st layer

mooklepticon, 2023-03-17:
Is there a known issue with PS 2.5.1 and Octolapse? Not Octoprint, that's fine, but Octolapse? If I have octolapse, then my print has an extruder issue after layer 1. It prints the layer 1 fine, then takes the snapshot, then moves back and will not extrude anything. The extruder just wiggles very slightly. It'll do the retract and reprime, but then not actually extrude anything. If I turn of octolapse, it prints fine.

Printer & Slicer - Ender 3 original, PS 2.5.1
Filament Material and Brand - PLA, unknown
Nozzle and Bed Temperature - 210, 60
Print Speed - 50
Nozzle Retraction Settings - 6

#### Version of Octolapse
 
Octolapse Version: 
Octolapse 0.4.1
Settings Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 

    OctoPrint 1.8.7 Python 3.7.3 OctoPi 0.18.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Print with Cura 5.3, it works fine.
2. Print with PS 2.5, it stops extruding after layer 1.
3. Print with PS 2.5 and Octolapse disabled, it prints fine

#### What should have happened?
Should have kept printing layers 2 and beyond.

#### What happened instead?
Extruder wiggles after taking octolapse snapshot. Still retracts and reprimes, but no actual extrusion during print moves.

#### Operating System running OctoPrint and Octolapse

    OctoPrint 1.8.7 Python 3.7.3 OctoPi 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware Version: Marlin 2 bugfix

#### Browser and version of browser, operating system running browser

Browser: Firefox Win10, Firefox Android

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [yafic_v2 t210 f100 (PS) - Copy.txt](https://github.com/FormerLurker/Octolapse/files/11004085/yafic_v2.t210.f100.PS.-.Copy.txt)

#### Link to settings.json

Link to settings.json with all passwords removed: 
[settings.json.txt](https://github.com/FormerLurker/Octolapse/files/11004107/settings.json.txt)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: 
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/11004153/plugin_octolapse.log)

#### Link to octoprint.log

Link to octoprint.log:  [octoprint.log](https://github.com/FormerLurker/Octolapse/files/11004150/octoprint.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: n/a

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

https://user-images.githubusercontent.com/19435974/225979208-eaa42ff6-e80f-40e0-9d32-8d4bdd93b831.mp4

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2023-04-02:
Sorry for the delay, but could you post your video to youtube or something?  It's not loading here.

Also, I looked at your settings, and your slicer type is set to cura > 4.2.  Please set it to 'automatic' and try again.  I would recommend using the default settings for your printer profile (for the ender 3), and reading [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration#if-you-are-using-cura-follow-these-steps) about getting cura to work with the 'automatic' settings.

E1Guason2024, 2023-12-22:
I got the same problem, testing i found when is stop printing.

In prusa or cura if i activate the zlift (prusa) or zhop (cura), in the second layer stop extruding, the extruder do the "wiggles"  commented by mooklepticon.

When desactivate z lift or z hop, the print go well.

Sorry for my english, not native.