FormerLurker/Octolapse #888: Cannot install Octolapse on Python 3.10.6

dl9sec, 2023-03-18:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.8.7 (via [paukstelis](https://github.com/paukstelis/octoprint_install) installation script)

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Install Octolapse via OctoPrint Plugin Manager

#### What should have happened?
Octolapse should be installed without errors.

#### What happened instead?
The Octolapse installation throws some errors (complaining there is something deprecated and that a "wheel" could not be created).
Installation finished, but Octolapse is unusable.
I am not sure if it has something to do with [this](https://github.com/FormerLurker/Octolapse/issues/787), but there is no remark ab out a "wheel" problem.

#### Operating System running OctoPrint and Octolapse

OS Name: Armbian (64 bit) on Odroid C2
Os Version: 23.02

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3S+ (not relevant)
Printer Firmware Version: 3.12.2 (not relevant)

#### Browser and version of browser, operating system running browser

Browser: Firefox (not relevant)
Browser OS: 111.0 (64 bit) (not relevant)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  [Octolapse_installation_OdroidC2_Armbian23.02_OP1.8.7_Py3.10.6](https://pastebin.com/Shv3cC6Y)

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

gald33, 2023-03-29:
> 

where is "... from URL" you refer to? can't find it, I'm using an updated octo4a installation.

dl9sec, 2023-03-29:
> > 
> 
> where is "... from URL" you refer to? can't find it, I'm using an updated octo4a installation.

See https://docs.octoprint.org/en/master/bundledplugins/pluginmanager.html
Figure 19

Regards, Thorsten

maclaine-san, 2023-03-31:
@dl9sec Thank you, i was having trouble installing octolapse on my octo4a app as well. This solved it for me.

FormerLurker, 2023-04-02:
All, this issue should be fixed with the latest release.  You might not see it yet in the plugin manager.  Please let me know if you continue to have problems, and SO SORRY for the delays!!!!!

dl9sec, 2023-04-03:
Unfortunately no.
The wheel for octolapse still cannot be build.

The last message is:
`DEPRECATION: Octolapse was installed using the legacy 'setup.py install' method, because a wheel could not be built for it. pip 23.1 will enforce this behaviour change. A possible replacement is to fix the wheel build issue reported above. Discussion can be found at https://github.com/pypa/pip/issues/8368`

Need more log infos?

Regards, Thorsten

gald33, 2023-04-03:
> > > 
> > 
> > 
> > where is "... from URL" you refer to? can't find it, I'm using an updated octo4a installation.
> 
> See https://docs.octoprint.org/en/master/bundledplugins/pluginmanager.html Figure 19
> 
> Regards, Thorsten

Thanks a lot! I managed to install this way. I did get a bunch of errors but the installation of the fork ended up successfully.
I ran many tests and octolapse works great now!

KlutchKyle, 2023-04-03:
Still having problems installing due to Pillow versioning. I also have the problem with the fork that @dl9sec mentions

FormerLurker, 2023-04-04:
I will add the comma and will let you know when I push it.

dl9sec, 2023-04-04:
That would be great, thanks :-)

FormerLurker, 2023-04-04:
So, there was a bit more work to do in order to get Pillow working.  The versions listed weren't 100% compatible with Python 11, and the import has changed.  I bumped the requirements to Pillow >= 9.3, <11 (see [this link](https://pillow.readthedocs.io/en/stable/installation.html)), added imports for both namespaces (pil and PIL).  I have tested this in Python 3.9.2 (latest octopi) and Python 3.11.2 (windows).  So far so good.

Try installing from the plugin manager by clicking 'get more' and using this url:  https://github.com/FormerLurker/Octolapse/archive/refs/heads/devel.zip

like so:

[image]

Please let me know what kind of system you are installing this on (raspberry pi running OctoPi, Windows, Ubuntu, etc...).  Just helps me feel confident putting this on the main branch later.

Thanks!

geekcouplettv, 2023-04-04:
I also have problem installing it on my Octoprint new setup on Windows Server 2019

[Installing plugin Octolapse from ht.txt](https://github.com/FormerLurker/Octolapse/files/11152337/Installing.plugin.Octolapse.from.ht.txt)

dl9sec, 2023-04-05:
Great stuff, thank you very much!
Installation was running without problems on my Odroid-C2 with Armbian 23.02.2 (Jammy) and Python 3.10.6.
Octolapse  v0.4.3+u.d8b2b5b seems to work at least with all the settings stuff (didn't make a print and timelapse yet).

Regards, Thorsten

FormerLurker, 2023-04-05:
@dl9sec,

Glad to hear it's working!  

Off topic - I have a question if you have a sec:  How is your Odroid working for you?  Many of my projects are suffering for a lack of affordable Raspberry Pis, and I've been looking for an alternative.

Also, fyi, I looked at your site, and learned that I lived close to you a long while ago (if you were there in 2002ish) during a year abroad at Stuttgart Uni.  I visited again a few years back, and things changed a lot!  Still a beautiful part of the country, and I look forward to visiting again!

MFG

dl9sec, 2023-04-05:
@FormerLurker

"The world ia a village" :-D Agree, beautiful area here.

The Odroid-C2 is working really great. It's a mature device, several years old, but is has a better performance than the (currently very expensive) RasPi 3B I used before. It runs an Armbiam OS (the original Hardkernel OS lacks a bit of support and up-to-date kernel and OS). Sometimes it is a bit tricky to get that RasPi-oriented stuff working, but that's not a problem of the Odroid itself.
So my personal conlusion is: If you get an up-to-date kernel/OS for an Odroid, it's a really good (maybe better and cheaper) alternative to the RasPi...

JDMorrison75, 2023-09-22:
> So, there was a bit more work to do in order to get Pillow working. The versions listed weren't 100% compatible with Python 11, and the import has changed. I bumped the requirements to Pillow >= [...]

I am having issues I have tried this current link I have tried to install the wheel, pillow etc through ssh. it always gets to installing the wheel and throws me errors. I am use octoprint 1.9.2  I have octoprint on a blu view 2 android phone.
[plugin_pluginmanager_console (1).log](https://github.com/FormerLurker/Octolapse/files/12704493/plugin_pluginmanager_console.1.log)