FormerLurker/Octolapse #889: Octolapse: Snaphots before and after completed print

PaWi20, 2023-03-22:
#### If this is a feature request describe it here
Most of the time, I use the "Smart - compatibility" or "Smart - Snap to Print" Triggers. Unfortunately, they don't take a snapshot before the print (empty print plate) and after the print, with all layers finished. I've seen timelapses where the last snapshot was taken after printing the shell surface of the print.
As far as I understood, its also possible to use G-Code commands (@OCTOLAPSE TAKE-SNAPSHOT) to trigger snapshot and stabilization. For this I have to adjust my G-code with the trigger commands before/after the print as well at layer changes. But I also must use the "Gcode" Trigger.

It would be great to have the Smart triggers mentioned above in combination with the G-code command. In this case I only need to edit my start/end G-Code and can use all the smart triggers, that I really like.

Or maybe did I miss a setting that solves my problem to take snapshots before and after the print?

#### Version of Octolapse
Octolapse Version: 0.4.1

#### Version of OctoPrint
OctoPrint Version: 1.8.7

FormerLurker, 2023-04-02:
YES, I have thought about this.  I may even add the ability to take 'before' and 'after' print snapshots in addition to allowing @Octolapse commands to be used in any of the triggers.

FormerLurker, 2023-04-03:
I believe I have it working.  Try installing within the plugin manager using this link (get more->...from URL-> paste link -> click install button): https://github.com/FormerLurker/Octolapse/archive/refs/heads/devel.zip

Now you should find a new option within the 'smart trigger' settings (it will be set to 'enabled' by default):

[image]

Now, just put the @OCTOLAPSE TAKE-SNAPSHOT command wherever you want (start and end of print)!

Please let me know how it works.

PaWi20, 2023-04-04:
Hey there. Thank you very much for your great work, especially for this complete plugin! I'm going on vacation in two days and I'll have to see if I can find the time to test. I'll come back to you!

3dflea, 2023-10-17:
Not sure if you've had feed back from a prusa slicer user, but I cannot get this new option to work.

Octopi 1.0 octoprint 1.9.3 new camera stack. 
Verbose Gcode selected.
Added custom command @OCTOPRINT TAKE-SNAPSHOT within start Gcode before priming line 

Selecting the option for smart triggers - compatibility and tick the Allow snapshot command option.

The first snapshot happens roughly half way through the first layer.

PaWi20, 2023-10-21:
No, I didn't. But I tested a timelapse with the new camera stack yesterday and it worked as expected (Cura).