FormerLurker/Octolapse #890: Help installing octolapse on octo4a

gald33, 2023-03-27:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.8.7

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

I also posted this issue on octo4a's github [here](https://github.com/feelfreelinux/octo4a/issues/397)
Hi, I'm running into a problem installing octolapse on octo4a. I tried a bunch of solutions I found here and there.
I installed a clean version of android and octo4a and then ran commands via ssh to fix a missing wheel (I think):

```
localhost:~# pip3 install --upgrade setuptools wheel tornado --no-cache-dir
localhost:~# curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s
localhost:~# unzip octoprint.zip
localhost:~# cd Octo* && pip3 install .
```

I did install some other plugins to see that it isn't a general issue.

I'm getting the following error:

```
Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpg2gbgehx/Octolapse-master.zip --no-cache-dir
Processing /tmp/tmpg2gbgehx/Octolapse-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'error'
error: subprocess-exited-with-error
× python setup.py egg_info did not run successfully.
│ exit code: 1
╰─> [10 lines of output]
Argon2 passlib backend is not available, not using it for password hashing
/tmp/pip-req-build-gj2xxwk9/setup.py:58: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
if LooseVersion(octoprint.server.VERSION) < LooseVersion("1.4"):
error in Octolapse setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after version specifier)
pillow >=6.2.0<7.0.0
~~~~~~~^
likely tags: v0.4.1
picking v0.4.1
got version from expanded keyword {'version': 'v0.4.1', 'full-revisionid': 'cec243c65a979e8bbe90f09fc5aec6d0844d2d3b', 'dirty': False, 'error': None, 'date': '2020-12-04T18:35:29-0600'}
Found packages: {'octoprint_octolapse_setuptools', 'octoprint_octolapse.test', 'octoprint_octolapse'}
[end of output]
note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed
× Encountered error while generating package metadata.
╰─> See above for output.
note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```
This is the contents of the log file mentioned in the last line:

```
localhost:~/.octoprint/logs# cat plugin_pluginmanager_console.log
2023-03-27 13:15:40,167   /usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp0fw6ht4o/Octolapse-master.zip --no-cache-dir
2023-03-27 13:15:43,405 > Processing /tmp/tmp0fw6ht4o/Octolapse-master.zip
2023-03-27 13:15:44,408 > Preparing metadata (setup.py): started
[... 83 lines cut ...]

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

sv0o, 2023-03-28:
have you tried the a-c-m fork mentioned here yet?  https://github.com/FormerLurker/Octolapse/issues/888#issue-1630500080

FormerLurker, 2023-04-02:
This should be fixed by the latest release, which will run on python 3.10/3.11.  So sorry for the delays, and let me know if there are issues.

gald33, 2023-04-03:
> have you tried the a-c-m fork mentioned here yet? [#888 (comment)](https://github.com/FormerLurker/Octolapse/issues/888#issue-1630500080)

I'm glad the original is fixed, I haven't tested it yet. I did manage to install and test the fork you referenced and it solve my problem. Thanks a lot!

closed 2023-04-03 by gald33

Cedric9602, 2023-04-30:
Hi all :)
I have tried to install OctoLapse the last days but was not successfull at it.. 
If i try to install it directly via the plugin repository I receive the folloing error:
× python setup.py egg_info did not run successfully.
│ exit code: 1

Via the link https://github.com/FormerLurker/Octolapse/archive/refs/heads/devel.zip I get the following errors 
× python setup.py bdist_wheel did not run successfully.
│ exit code: 1
...
compilation terminated.
error: command 'gcc' failed with exit code 1

I am really new to OctoPrint and did search really long to help myself but I am out of ideas now.. 
I already installed a ton of packages via the web terminal of Octo4a:
apk add cmake
apk add py3-setuptools
apk add py3-wheel
apk add py3-pip
apk add py3-pillow
apk add gcc
apk add build-base
apk add alpine-sdk
apk add clang
apk add py3-clang

pip3 install clang
pip3 install --upgrade pip setuptools wheel pillow tornado
pip3 install libevent-dev
pip3 install ez_setup

Octoprint version is 1.8.7.
If any more information is needed please tell me :)
I would really like to take some cool OctoLapse timelapse recordings 
Thanks in advance!