FormerLurker/Octolapse #899: [Request] Add a choice to run the "Before Render Script" even if Rendering is set to Disabled

dssamusaran, 2023-04-27:
#### If this is a feature request describe it here

When the Rendering is set to 'Disabled' we should be able to still run at least the 'Before Render Script'.

When using the 'Deferred Download' documentation ([See here](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-DSLR-With-Deferred-Download)), the pictures are downloaded from the DSLR through this 'Before Render Script'.

It works fine in most cases but if you set the Rendering to 'Disabled' the script is not ran at all and you end up with a .zip file without the snapshots and containing only the folder structure, metadatas and such.

I also tried putting the deferred download script in the 'After Print End Script' section but it does not work either. It looks like the empty .zip is already created before this section is ran.

Maybe a setting in the Rendering Profile, Timelapse Generation part letting us untick Enabled and having a second box "Still run scripts while disabled" could be a way to implement this. 

This would be useful for people doing the rendering in external softwares.

#### Version of Octolapse
 
Octolapse Version: v0.4.2

#### Version of OctoPrint

OctoPrint Version: 1.8.7

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2023-05-13:
> I also tried putting the deferred download script in the 'After Print End Script' section but it does not work either. It looks like the empty .zip is already created before this section is ran.

Yeah, it probably won't work exactly correctly here, since that script requires a payload that is generated in the rendering routine.  I'll look into this.

FormerLurker, 2023-05-13:
So, it turns out this was pretty easy to add.  I haven't tested it because my DSLR isn't configured (takes a while and just rebuilt my system).  Please give it a try and let me know.  You can install from the devel branch in the plugin manager (get more, ..from url) with this link: https://github.com/FormerLurker/Octolapse/archive/refs/heads/devel.zip

dssamusaran, 2023-05-16:
Hi there.

It's working perfectly fine now, thanks for the support!

Do you think this will make it to the next release or should I stick with the devel release?

Thanks.

FormerLurker, 2023-05-16:
Oh, it will be in there.  The devel version you are running is perfectly safe (just 1 line of code has changed) to use, and you should still be notified of the next release.

Glad to be of assistance!  Feel free to close the issue if you feel it's been resolved :)

dssamusaran, 2023-05-16:
Great!

This is indeed perfectly resolved, I'm currently editing the timelapse done thanks to your fix :)

closed 2023-05-16 by dssamusaran