FormerLurker/Octolapse #900: Unable to install OctoLapse on Octo4a with OctoPrint Version 1.8.7

Cedric9602, 2023-05-01:
#### What were you doing when the problem occurred
I have tried to install OctoLapse the last days but was not successfull at it..
If i try to install it directly via the plugin repository I receive the folloing error:
× python setup.py egg_info did not run successfully.
│ exit code: 1

Via the link https://github.com/FormerLurker/Octolapse/archive/refs/heads/devel.zip I get the following errors
× python setup.py bdist_wheel did not run successfully.
│ exit code: 1
...
compilation terminated.
error: command 'gcc' failed with exit code 1

I am really new to OctoPrint and did search really long to help myself but I am out of ideas now..
I already installed a ton of packages via the web terminal of Octo4a:
apk add cmake
apk add py3-setuptools
apk add py3-wheel
apk add py3-pip
apk add py3-pillow
apk add gcc
apk add build-base
apk add alpine-sdk
apk add clang
apk add py3-clang

pip3 install clang
pip3 install --upgrade pip setuptools wheel pillow tornado
pip3 install libevent-dev
pip3 install ez_setup

#### What should have happened?
Install should work 

#### What happened instead?
Mentioned errors 

#### Operating System running OctoPrint and Octolapse
Octo4a 1.2.4 on NVIDIA Shield with Android 7 

#### Browser and version of browser, operating system running browser
Microsoft Edge

If any more information is needed please tell me :)
I would really like to take some cool OctoLapse timelapse recordings

Thanks in advance!

9cento, 2023-05-13:
It's not only the typo in required pillow version, gcc in Alpine Linux 3.17 (base of Octo4a) won't compile pillow and can't be downgraded

FormerLurker, 2023-05-13:
@9cento, have you tried installing the devel build?  I just promoted this to master and added a release (check for updates in the Software Update plugin in OctoPrint).  I've gotten several reports of success with Octo4a, and am hoping this isn't something new.

9cento, 2023-05-13:
@FormerLurker I tried the devel branch yesterday if I remember correctly, without success, but I will try the new main release asap and thank you for your celery response

9cento, 2023-05-13:
@FormerLurker Sadly, still similar issues.
I tried the new OctoLapse-0.4.3 with Octo4a-1.2.4 (who had success with the installation of OctoLapse probably runs an older version of Octo4a but I can't downgrade since there's no Android 11 compatibility in the previous releases) and Octoprint-1.8.7, all of this runs on top of an Alpine Linux virtual environment that enables Octo4a to work by default.
The Alpine Linux version is 3.17.1, the build-base package is 0.5-r3 and the gcc version is 12.2.1_git20220924-r4 (which as far as I can tell is the root cause of the problem since it won't compile anything as you will see in the log).
I hope you can find a solution and thanks for your effort, very appreciated.
Here's the full log straight from Octoprint itself: [Log on Pastebin](https://pastebin.com/fV2fwJqx) since it is very long.

FormerLurker, 2023-05-13:
Did you go to the settings in the octo4a app and check the box that says "install plugin extras" ?

9cento, 2023-05-13:
@FormerLurker yes I did and it went smooth. I hope it's not an incompatibility between Android 11 and Alpine Linux 3.17 because that would be a major pain in the a$$

FormerLurker, 2023-05-13:
I will look at the log when I have some time, but my guess is that there is some issue with the compiler (maybe it isn't installed?  No clue)

9cento, 2023-05-13:
@FormerLurker thanks for your time. As for the compiler I verified and it's there, it just doesn't work as expected, I also force-reinstalled it but still same problem

FormerLurker, 2023-05-13:
I did find this during a brief scan (am mobile ATM): gcc: fatal error: cannot execute 'cc1plus': execvp: No such file or directory

9cento, 2023-05-13:
@FormerLurker also checked it and all ok, checked even the PATH environment variables and it's all on spot, I'm pretty much lost at this point

FormerLurker, 2023-05-13:
You could try compiling the c manually and see how that goes.  If you get a binary out of it, I can walk you through a manual install.  Worst case maybe we learn something?  I will spend some time researching tomorrow.

9cento, 2023-05-13:
@FormerLurker I'm afraid my knowledge stops right before that lol so I'll do my own research but I can't guarantee I will nail it

FormerLurker, 2023-05-13:
Understood.  Did you try this:

apk add g++

??

9cento, 2023-05-13:
@FormerLurker yep, it's in place

FormerLurker, 2023-05-13:
Ok.  I am on discord with the same username.  Hit me up tomorrow at around 11CT and we can debug.  In the meantime, see if you can try a manual compile.  I realize that is a big ask, but you may gain some additional insight that could prove useful.

9cento, 2023-05-14:
@FormerLurker uhm this is very strange: if I run as # `gcc helloworld.c -o helloworld` I get `-ash: gcc: Permission denied`, but if I run `/usr/bin/gcc helloworld.c -o helloworld` it compiles smoothly, the same applies to g++.
After this I double-checked my PATH and `/usr/bin` is indeed there. Finally I run `find / -name gcc 2>/dev/null` just to check for potential conflicting gcc packages and it printed the following:
`/usr/libexec/gcc`
`/usr/lib/gcc`
`/usr/bin/gcc`
It's worth to mention that `/usr/libexec` and `/usr/lib` are NOT in my PATH, frankly I don't know if it's supposed to be like that.

9cento, 2023-05-14:
> @FormerLurker uhm this is very strange: if I run as # `gcc helloworld.c -o helloworld` I get `-ash: gcc: Permission denied`, but if I run `/usr/bin/gcc helloworld.c -o helloworld` it compiles [...]

@FormerLurker ok partial progress: I managed to get gcc to work in the ssh shell by just prompting `gcc` instead of `/usr/bin/gcc` by adding `alias gcc='/usr/bin/gcc` to `etc/profile` (applied this very same dirty fix for g++), this way I was able to complile my example by just prompting `gcc helloworld.c -o helloworld`. Then I tried to reinstall again OctoLapse-0.4.3 via Octoprint's plugins manager by manually adding the new url, sadly it bitched again about `gcc: fatal error: cannot execute 'cc1plus': execvp: No such file or directory` (I guess I'll dig more into that now). Just for the sake of it I also tried to install the plugin via the ssh shell but same result, will report back if there will be any new findings

9cento, 2023-05-14:
> @FormerLurker ok partial progress: I managed to get gcc to work in the ssh shell by just prompting `gcc` instead of `/usr/bin/gcc` by adding `alias gcc='/usr/bin/gcc` to `etc/profile` (applied this [...]

@FormerLurker I think we are almost there: I managed to fix `gcc: fatal error: cannot execute 'cc1plus': execvp: No such file or directory` with a symlink, now I'm getting this:
`gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -DIS_PYTHON_EXTENSION=1 -I/usr/include/python3.10 -c octoprint_octolapse/data/lib/c/extruder.cpp -o build/temp.linux-armv8l-cpython-310/octoprint_octolapse/data/lib/c/extruder.o -O3 -std=c++11`
`In file included from octoprint_octolapse/data/lib/c/extruder.cpp:2:`
      `octoprint_octolapse/data/lib/c/logging.h:3:10: fatal error: string: No such file or directory`
          `3 | #include <string>`
            `|          ^~~~~~~~`
      `compilation terminated.`
      `error: command 'gcc' failed with exit code 1`
As you can see now the problem is no longer related to the actual execution capability of the gcc command in general but it rather resides in how that manages the code which is trying to (finally!) compile.
So I was thinking that **maybe** if `g++` is used to compile instead of `gcc` the installation could go trough, correct me if I'm wrong, but I don't know how to force such behavior (messing with the source code is probably needed but sadly it's out of my scope).
That's all from me for now, do your tricks and let me know once you'll have time, no need to hurry obviously 🤗

9cento, 2023-05-14:
> @FormerLurker I think we are almost there: I managed to fix `gcc: fatal error: cannot execute 'cc1plus': execvp: No such file or directory` with a symlink, now I'm getting this: `gcc [...]

@FormerLurker Yet another update: managed to force the use of `g++` instead of `gcc` but it didn't change a thing, same error.

FormerLurker, 2023-05-14:
@9cento, looks like there is some progress!  How did you 'force' the use of g++?  Did you do this maybe:

```export CXX=g++```

before installing?

Let's try a c++ file instead of a c file and see if you can compile that with g++.  Create a file named hello_world.cpp and fill it with the following content:

```
#include <iostream>
#include <string>

int main() {
    std::string greeting = "Hello, World!";
    std::cout << greeting << std::endl;
    return 0;
}
```
Then compile it with the following command: ```g++ -o hello_world hello_world.cpp```

And run it (if the compilation was successful) with this: ```./hello_world```

You'll notice that I included <string> here, which is failing during the Octolapse install.

Fyi, I'm online (FormerLurker#5525) if you want to chat instead.  I may delete this line, so copy the username :)

9cento, 2023-05-14:
@FormerLurker username copied for future reference, but for now I'd prefer to talk here even if it's less pratical, just in case someone will find himself in the same spot with this obscure issue.
To compile with `g++` I temporarily forked the repo and put `self.compiler.set_executable("compiler_so", ["g++"])` in `[setup.py](http://setup.py/)`, I don't know how but it did the trick (thanks ChatGPT lol) because when it printed the usual error it cited `g++` instead of `gcc`.
As for the `hello_world.cpp` file, it compiled and run no problem.

FormerLurker, 2023-05-14:
So, for some reason, the compiler is unable to locate the standard libraries, but only when running it through octoprint.  V strange.  Perhaps this is masking some other issue?

When you installed g++, did you include the python dev libraries:  ```apk add g++ python3-dev```  I'm guessing there would be a different error message than a missing string library....

Have you tried running setup.py for Octolapse manually?  You should be able to activate the virtual environment and run setup.py yourself.  [This guide](https://docs.octoprint.org/en/master/plugins/gettingstarted.html#saying-hello-how-to-make-the-plugin-actually-do-something) for getting started with plugin development should explain some of the nuance.

9cento, 2023-05-14:
> the compiler is unable to locate the standard libraries, but only when running it through octoprint

Nope, same thing from the ssh shell.

> When you installed g++, did you include the python dev libraries: `apk add g++ python3-dev`

Yep, that's all on place.

> Have you tried running setup.py for Octolapse manually? You should be able to activate the virtual environment and run setup.py yourself.

Yeah same error sadly.

yaukari2, 2023-06-01:
was a solution found?

9cento, 2023-06-02:
> was a solution found?

Personally, I wasn't able to find a workaround and gave up on the thing. You may want to ask @FormerLurker if there are any news on that.

FormerLurker, 2023-06-02:
I just need to replicate the issue, which probably means purchasing some hardware.  If anyone has any suggestions or steps to reproduce the issue on some common hardware, that would be helpful.

I am also considering submitting the octolapse C++ extensions to pypi so it can be installed via PIP, or creating some GitHub runner to make precompiled binaries for all of the common platforms.  Anyone with experience in doing either of these would be my hero, as the compilation steps have always been an issue for me.  I do have CI for the Arcwelder c++ using cmake, so it is possible, and I believe that GitHub now supports some arm compilation.

9cento, 2023-06-02:
@FormerLurker ok good news: I can confirm that the issue is replicable on Android Studio's AVD emulator with Android 11 so no physical hardware is needed to troubleshoot it.
Also, I'm almost sure that it's not even a cpu architecture/instructions related problem since during octo4a installation it baked an x86 bootstrap for Alpine Linux while so far we only tested the thing on Arm based devices.
Now it's up to you my friend, my knowledge stops here so sadly I can't probably help you any further, but anyway feel free to hit me up here or on Discord for whatever!

9cento, 2023-06-13:
@FormerLurker sorry to bother you, I just wanted to know if there are any updates on this issue

FormerLurker, 2023-06-16:
Thanks for the nudge!  Unfortunately, not yet.  I will try to make progress this weekend.  Please keep pestering me.

9cento, 2023-06-16:
> Please keep pestering me.

Man I went radio silent for like a month, nobody is pestering you

chis54, 2023-06-20:
> 

I posted in another issue thread (https://github.com/FormerLurker/Octolapse/issues/911#issuecomment-1595942980 and log posted with comment) and the issue may be related to what you were discussing here. I am running Octo4a v1.2.5 on a Pixel 5 with Android 13. Other plugins install and run just fine.

calif94577, 2023-08-13:
I think I am having the same issue. Here is the error, too long to comment here. [https://pastebin.com/NXpUAbPi](url)

calif94577, 2023-08-14:
> > Please keep pestering me.
> 
> Man I went radio silent for like a month, nobody is pestering you

I don't think that was sarcasm I think he legit wanted you to stay on top of him so he doesn't forget.

@FormerLurker Any progress? Saw on the issue linked 3 comments above this that someone made a fork that might be working? No sure. Thanks!

BaTaTaAdb, 2023-08-15:
Hey. I've been researching and found that by installing an older version of octolapse I got it to **install** on my huawei P40 lite with the latest octo4a version. Found it on [#888](https://github.com/FormerLurker/Octolapse/issues/893#issuecomment-1508792881).

install it using the following link: [https://github.com/a-c-m/Octolapse/archive/refs/heads/master.zip](https://github.com/a-c-m/Octolapse/archive/refs/heads/master.zip)

**Haven't tested it in a print yet!!**

shashwat001, 2023-12-28:
After 3 days of effort I am able to install it. Enable ssh and install the package as follows.
`pip install "https://github.com/FormerLurker/Octolapse/archive/master.zip"`

It will start with following error:
> gcc: fatal error: cannot execute 'cc1plus': execvp: No such file or directory

Resolution is to add `cc1plus` to path. Find the location of `cc1plus` using the command
`/usr/bin/gcc --print-prog-name=cc1plus`
and either add the path to PATH envvar, or symlink cc1plus to an existing path.

Next error would be

> gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC [...]

Its because installation is not able to find `string` in include dirs. This command works directly but somehow not working with `pip`. The solution is to add include directories in `setup.py` file. This will require extracting the directory say `/tmp/Octolapse-master`.

Modify `setup.py` and add `include_dirs` to `cpp_gcode_parser` Extension variable
```
cpp_gcode_parser = Extension(
    'GcodePositionProcessor',
    sources=plugin_ext_sources,
    language="c++",
    include_dirs=['/usr/include/c++/12.2.1', '/usr/include/c++/12.2.1/armv6-alpine-linux-musleabihf']
)
```
If `/usr/include/c++/12.2.1/armv6-alpine-linux-musleabihf` path is not same, search for `c++config.h` file location and use that path.
`find / -name c++config.h 2>/dev/null`

Install using 
`pip install file:///tmp/Octolapse-master`

sanjiv204, 2024-12-30:
> Did you go to the settings in the octo4a app and check the box that says "install plugin extras" ?

It worked for me. This happened to me after the update and a clean reinstall of octo4a on Samsung m30