FormerLurker/Octolapse #903: STOP-SNAPSHOTS in gcode but still moving nozzle to center

jacopomaroli, 2023-05-07:
#### If this is a feature request describe it here

N/A

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.8.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: no

#### What were you doing when the problem occurred

1. I'm injecting the following start gcode through cura at the beginning of every print
```
M117 Cleaning the nozzle
@OCTOLAPSE STOP-SNAPSHOTS
G90 ; absolute XYZ mode
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z2.0 F5000.0 ;Move above start position
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
; ==> octolapse moves the nozzle to the middle of the plate here <==
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ; reset extruder
G1 Z2.0 F3000 ; move z up little to prevent scratching of surface
@OCTOLAPSE START-SNAPSHOTS
```
2. Octolapse trigger set to "smart - compatibility"
3. I start the print
4. nozzle moves to the middle of the plate when it shouldn't (see comment `octolapse moves the nozzle[...]` in prev snippet)

#### What should have happened?
when I use @OCTOLAPSE STOP-SNAPSHOTS it should not move the nozzle to the middle of the plate 

#### What happened instead?
nozzle moved to the middle of the plate

#### Operating System running OctoPrint and Octolapse

OS Name: Linux raspberrypi
Os Version: 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

FormerLurker, 2023-05-13:
I will test this out ASAP.  FYI, you are a couple versions behind (0.4.3 just dropped), so if you have a chance to update and re-test on the latest version, I'd appreciate that!

FormerLurker, 2023-05-13:
OH, please provide your Gcode file (at least the first couple layers), and an export of your settings.  See [this wiki entry](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Reporting-An-Issue) for details on how to accomplish that.  Just follow the [Include Your settings.json File](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Reporting-An-Issue#include-your-settingsjson-file) and [Include Your Gcode File](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Reporting-An-Issue#include-your-gcode-gile) instructions (excuse the typo).

jacopomaroli, 2023-05-17:
Hey there, thanks for looking into this :)
please find here my [gcode](https://gist.github.com/jacopomaroli/d9785218d8b9455ae4f7684b4e090056) and my [settings.json]( https://gist.github.com/jacopomaroli/d6c2bf58c68b930c1aab656cd353f837)

I just updated everything to the latest versions; just give me a few mins to try again. I'll update this comment with the outcome