FormerLurker/Octolapse #911: Installing in octo4a fails (compilation error)

philipp, 2023-06-03:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: fails to install "master.zip" on 2023-06-03

#### Version of OctoPrint

OctoPrint Version: 1.9.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No (I don't think I can, during the plugin installation process)

#### What were you doing when the problem occurred

1. Open OctoPrint Settings -> Plugin Manager
2. Click "Get More"
3. Click "Install" button next to OctoLapse

#### What should have happened?
It should have installed the latest OctoLapse

#### What happened instead?
OctoPrint tried multiple times to install, but failed.

#### Operating System running OctoPrint and Octolapse

OS Name: Android
Os Version: 10

octo4a version 1.2.4

The rest of this template seems not relevant to an install-time problem.
But what is presumably relevant is the error from the install log.
Plenty of warnings, but just this one error section:

from octoprint_octolapse/data/lib/c/gcode_position_processor.cpp:24:
octoprint_octolapse/data/lib/c/gcode_position_processor.cpp: In function 'bool ParseStabilizationArgs(PyObject*, stabilization_args*, PyObject**, PyObject**)':
octoprint_octolapse/data/lib/c/gcode_position_processor.cpp:1404:24: error: converting to 'bool' from 'std::nullptr_t' requires direct-initialization [-fpermissive]
1404 |                 return NULL;

I note that all the other "if" conditions return "false", but this particular one returns NULL
It's been too long for me to have an opinion on the specific casting error/restriction this is complaining about, but maybe it's a simple fix?

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

chis54, 2023-06-18:
I'm having what might be the same issue. Fresh installation of Octo4a v1.2.5. Octolapse plugin installation fails every time. 
Here is the log: 
[octolapseinstall.log](https://github.com/FormerLurker/Octolapse/files/11780589/octolapseinstall.log)
Installing other plugins seem to work just fine.

philipp, 2023-06-26:
I thought I had tried that.  I guess not.  Notably, it seems to have successfully installed from that fork, even though the log of the install showed multiple errors (all related to Pillow)
Now of course it's prompting me to update to the latest (4.3 from this 4.1+commit version, and the changelogs suggest I might actually want to, since I use SuperSlicer)
But I think I'll just try it out this way first, and see what happens.  Must. Resist. Pressing "Update now"!

philipp, 2023-06-30:
I've forked the current "master" and made a single change, reflecting what I described in this issue.  I changed `NULL` on line 1404 of `octoprint_octolapse/data/lib/c/gcode_position_processor.cpp` to `false`
And with that change, the installation succeeds!
I've created a Pull Request, but until that gets attention from @FormerLurker you can try installing from https://github.com/philipp/Octolapse/archive/refs/heads/philipp-issue-911.zip and see if it behaves for you.
I'm an OctoLapse noob, so unless it goes horribly wrong, I'm not sure I'd know if there were subtle problems introduced :-)

Obagunceiro, 2023-07-03:
> I've forked the current "master" and made a single change, reflecting what I described in this issue. I changed `NULL` on line 1404 of `octoprint_octolapse/data/lib/c/gcode_position_processor.cpp` [...]

That worked flawlessly!! thanks so much!

gglockner, 2023-07-24:
@philipp : Thanks for your help on this. I do get past the g++ error, but I get a different error when I build the wheel:

```
WARNING: Built wheel for Octolapse is invalid: Metadata 1.2 mandates PEP 440 version, but 'philipp-issue-911' is not
ERROR: Could not build wheels for Octolapse, which is required to install pyproject.toml-based projects
Building wheel for Octolapse (setup.py): finished with status 'done'
Created wheel for Octolapse: filename=Octolapse-philipp_issue_911-cp310-cp310-linux_aarch64.whl size=3147853 sha256=ad506701b889fb9205d3c8a8a849394eaa15b882a6a77d8961e04c79eb1892dc
Stored in directory: /tmp/pip-ephem-wheel-cache-cjslys02/wheels/c5/69/a0/d968101b1a55c091a2bdbd012f4ae795484bcb9509bb742c38
Failed to build Octolapse
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

philipp, 2023-07-26:
> @philipp : Thanks for your help on this. I do get past the g++ error, but I get a different error when I build the wheel:

@gglockner: No idea what that might mean, but I do notice that there's a newer version of Octoprint out now.  I'm still on 1.9.1, and that's what I was successful with.  Are you using 1.9.2?  Maybe there's some difference that breaks something else.

HaraldBluet00th, 2023-08-06:
> I've forked the current "master" and made a single change, reflecting what I described in this issue. I changed `NULL` on line 1404 of `octoprint_octolapse/data/lib/c/gcode_position_processor.cpp` [...]

Thank you for this - hours of searching brought me here and its  the only thing that has worked so far.

JDMorrison75, 2023-09-24:
I am using 1.9.2 and every time I get to building the wheel I get an error and it fails to install.
[plugin_pluginmanager_console (2).log](https://github.com/FormerLurker/Octolapse/files/12709284/plugin_pluginmanager_console.2.log)