FormerLurker/Octolapse #912: Server error while trying to save webcam settings

christopherfujino, 2023-06-13:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.3

OctoPrint Version: 1.9.0

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Navigate to "Octolapse" tab
2. Click edit pencil icon next to webcam (Logitech C920)
3. Without making changes, click "Save"

#### What should have happened?
Settings should have been saved (since no changes made, effectively nothing)

#### What happened instead?
An error alert pops up:

> Error Saving Camera Settings
> Status: error. Error: INTERNAL SERVER ERROR

#### Operating System running OctoPrint and Octolapse
OS Name: Octopi 
Os Version: 0.18.0

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:

(This doesn't look particularly interesting, see instead octoprint.log)

```
2023-06-13 00:59:55,709 - octolapse.camera - INFO - Applying all webcam image settings for the Logitech C920 camera.
2023-06-13 00:59:55,721 - octolapse.camera - DEBUG - Changing Brightness to 128 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963776&group=1&value=128
2023-06-13 00:59:55,762 - octolapse.camera - DEBUG - Changing Contrast to 128 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963777&group=1&value=128
2023-06-13 00:59:55,777 - octolapse.camera - DEBUG - Changing Saturation to 128 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963778&group=1&value=128
2023-06-13 00:59:55,797 - octolapse.camera - DEBUG - Changing White Balance Temperature, Auto to 1 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963788&group=1&value=1
2023-06-13 00:59:55,810 - octolapse.camera - DEBUG - Changing Gain to 0 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963795&group=1&value=0
2023-06-13 00:59:55,821 - octolapse.camera - DEBUG - Changing Power Line Frequency to 2 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963800&group=1&value=2
2023-06-13 00:59:55,833 - octolapse.camera - DEBUG - Changing White Balance Temperature to 4000 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963802&group=1&value=4000
2023-06-13 00:59:55,844 - octolapse.camera - DEBUG - Changing Sharpness to 128 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963803&group=1&value=128
2023-06-13 00:59:55,855 - octolapse.camera - DEBUG - Changing Backlight Compensation to 0 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=9963804&group=1&value=0
2023-06-13 00:59:55,865 - octolapse.camera - DEBUG - Changing Exposure, Auto to 3 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=10094849&group=1&value=3
2023-06-13 00:59:55,878 - octolapse.camera - DEBUG - Changing Exposure (Absolute) to 250 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=10094850&group=1&value=250
2023-06-13 00:59:55,891 - octolapse.camera - DEBUG - Changing Exposure, Auto Priority to 0 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=10094851&group=1&value=0
2023-06-13 00:59:55,902 - octolapse.camera - DEBUG - Changing Pan (Absolute) to 0 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=10094856&group=1&value=0
2023-06-13 00:59:55,913 - octolapse.camera - DEBUG - Changing Tilt (Absolute) to 0 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=10094857&group=1&value=0
2023-06-13 00:59:55,924 - octolapse.camera - DEBUG - Changing Focus (absolute) to 0 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=10094858&group=1&value=0
2023-06-13 00:59:55,935 - octolapse.camera - DEBUG - Changing Focus, Auto to 1 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=10094860&group=1&value=1
2023-06-13 00:59:55,947 - octolapse.camera - DEBUG - Changing Zoom, Absolute to 100 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=10094861&group=1&value=100
2023-06-13 00:59:55,958 - octolapse.camera - DEBUG - Changing JPEG quality to 50 for Logitech C920.  Request: http://127.0.0.1:8080/?action=command&dest=0&plugin=0&id=1&group=3&value=50
```

#### Link to octoprint.log
This is I believe the real error:

octoprint.log:

```
AttributeError: type object 'Retry' has no attribute 'BACKOFF_MAX'
2023-06-13 00:53:20,333 - octoprint - ERROR - Exception on /plugin/octolapse/saveWebcamSettings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1590, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1613, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_login/utils.py", line 290, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 1298, in save_webcam_settings
    success, error = camera.CameraControl.apply_camera_settings([camera_profile])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/camera.py", line 209, in apply_camera_settings
    thread.join(requests.packages.urllib3.util.retry.Retry.BACKOFF_MAX)
```

Note this was already filed upstream on octoprint 3 weeks ago, but closed: https://github.com/OctoPrint/OctoPrint/issues/4811

christopherfujino, 2023-06-13:
Here's the line: https://github.com/FormerLurker/Octolapse/blob/8d53c91069943b9b4d78498c2fc225b3d6d04cb8/octoprint_octolapse/camera.py#LL209C18-L209C18

```python
                thread.join(requests.packages.urllib3.util.retry.Retry.BACKOFF_MAX)
```

christopherfujino, 2023-06-13:
It appears I have version of `requests` v2.31.0 -> `urllib3` v2.0.3.

It looks like package urllib3 got rid of the `urllib3.util.retry.Retry.BACKOFF_MAX` static in https://github.com/urllib3/urllib3/commit/f69b1c89f885a74429cabdee2673e030b35979f0, allowing it to be configurable. I think the fix here (assuming all clients are using a new enough version of urllib3) would be to use `requests.packages.urllib3.util.retry.Retry.backoff_max`

christopherfujino, 2023-06-13:
FYI @beepboop1234

beepboop1234, 2023-06-13:
> FYI @beepboop1234

Thanks! It looks like you are seeing all of the same errors I am. If I can give any info or do any testing that would help fix this, just let me know. 

I noticed any changes I make seem to stick despite the errors so I've just been dismissing the pop-ups. Would love to not get the errors though.

rickgode, 2023-12-23:
I'm a complete noob, but know just enough to get in trouble.  I found the camera.py file and changed the line from thread.join(requests.packages.urllib3.util.retry.Retry.BACKOFF_MAX) to thread.join(requests.packages.urllib3.util.retry.Retry.**DEFAULT**_BACKOFF_MAX)

and that seemed to do the trick

christopherfujino, 2023-12-24:
> I'm a complete noob, but know just enough to get in trouble. I found the camera.py file and changed the line from thread.join(requests.packages.urllib3.util.retry.Retry.BACKOFF_MAX) to thread.join(requests.packages.urllib3.util.retry.Retry.**DEFAULT**_BACKOFF_MAX)
> 
> and that seemed to do the trick

I think we can close this since https://github.com/FormerLurker/Octolapse/pull/913/files was merged

closed 2023-12-24 by christopherfujino