FormerLurker/Octolapse #917: Timelapse previews not generated with Pillow 10.0.0

CodedSakura, 2023-07-13:
#### Version of Octolapse
 
Octolapse Version: 0.4.3

#### Version of OctoPrint

OctoPrint Version: 1.9.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Install OctoPi (1.0.0) on a raspberry Pi (3B+)
2. Add Octolapse plugin to OctoPrint
3. Configure Octolapse
4. Start printing with Octolapse enabled

#### What should have happened?
Timelapse preview should display the last snapshot

#### What happened instead?
Timelapse preview says "No snapshots have been taken...", but the Sanpshot Count displays a non-zero number.
The timelapse also renders succesfully after print finishes.

#### Self diagnosis of the problem
Looking at the plugin_oclolapse.log file, at [line 2202](https://gist.github.com/CodedSakura/a89dc65a8812503ded1f6fdcca5c2314#file-plugin_octolapse-log-L2202) it is visible that Octolapse failed to generate a thumbnail
from the snapshot. This seems to be caused by PIL, as the error says `AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'`.

Python version: 3.9.2  
Pillow version: 10.0.0

`PIL.Image.ANTIALIAS` was incidentally removed in Pillow 10.0.0, but was already deprecated for a while.

This issue repeats for each time Octolapse tries to generate a preview. ([L2202](https://gist.github.com/CodedSakura/a89dc65a8812503ded1f6fdcca5c2314#file-plugin_octolapse-log-L2202), [L3560](https://gist.github.com/CodedSakura/a89dc65a8812503ded1f6fdcca5c2314#file-plugin_octolapse-log-L3560), [L4898](https://gist.github.com/CodedSakura/a89dc65a8812503ded1f6fdcca5c2314#file-plugin_octolapse-log-L4898), [L6198](https://gist.github.com/CodedSakura/a89dc65a8812503ded1f6fdcca5c2314#file-plugin_octolapse-log-L6198), [L7480](https://gist.github.com/CodedSakura/a89dc65a8812503ded1f6fdcca5c2314#file-plugin_octolapse-log-L7480))

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 1.0.0

#### Printer model & used firmware incl. version

Printer Model: _Not needed, internal python problem_
Printer Firmware Version: _Not needed, internal python problem_

#### Browser and version of browser, operating system running browser

Browser: _Not needed, internal python problem_
Browser OS: _Not needed, internal python problem_

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: _Not needed, internal python problem_

#### Link to settings.json

Link to settings.json with all passwords removed: _Not needed, internal python problem_

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  <https://gist.github.com/CodedSakura/a89dc65a8812503ded1f6fdcca5c2314>

#### Link to octoprint.log

Link to octoprint.log:  <https://gist.github.com/CodedSakura/0ffe7486151dbe59dc29534f4054deaa>

#### Link to contents of Javascript console in the browser

Link to javascript console output: _Not needed, internal python problem_

#### Screenshots and/or videos of the problem:

Screenshot/Video Links:  
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

IntenseWiggling, 2023-07-14:
Encountered this as well, and was able to fix this by updating snapshot.py, replacing `Image.ANTIALIAS` with `Image.Resampling.LANCZOS`.

Ref:
https://pillow.readthedocs.io/en/stable/releasenotes/2.7.0.html#antialias-renamed-to-lanczos
https://github.com/python-pillow/Pillow/issues/6200

rsteckler, 2023-07-17:
confirmed the above change works.

Loeffelmaster, 2023-07-23:
Hi,
might be a dumb question, but what do I need to do? Install the sane plugin? Rename the Image.ANTIALIAS file to Image.Resampling.LANCZOS?
I'm a bit confused.
Thanks

Yours
Loeffel

xXxH3LP3RxXx, 2023-08-06:
> confirmed the above change works.

Same issue here, but i don't know how to make this change, any short tutorial?

Loeffelmaster, 2023-08-08:
> > confirmed the above change works.
> 
> Same issue here, but i don't know how to make this change, any short tutorial?

I found it while searching for something else. You need to edit a file on your Octoprint system.
I think it is /boot/octoprint.txt but I'm not sure and I can't have  look at the moment I have no access to my system and I can access it earliest in 7-14 days. In this file you need to change the entry accordingly. So you don't need to rename any file.

xXxH3LP3RxXx, 2023-08-08:
> I found it while searching for something else. You need to edit a file on your Octoprint system. I think it is /boot/octopi.txt but I'm not sure and I can't have look at the moment I have no access to my system and I can access it earliest in 7-14 days. In this file you need to change the entry accordingly. So you don't need to rename any file.

Thought i need to change the name in snapshot.py itself, but wasn't able to find it, also on octopi.txt i wasnt able to find anything, there are just some settings for the webcam [Octopi.txt](https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/boot/octopi.txt)

gnodabb, 2023-08-08:
> > confirmed the above change works.
> 
> Same issue here, but i don't know how to make this change, any short tutorial?

Fairly easy fix. If you open the octolapse log file you will see exactly where the error is located. Here is what I did.

1. ssh to octopi
2. cat or less plugin_octolapse.log (this will show where the error is occurring in the python code
3. using your favorite editor, change  Image.ANTIALIAS file to Image.Resampling.LANCZOS

```
$ ssh username@octopi
$ less ~/.octoprint/logs/plugin_octolapse.log
```

you should see something like this:
```
YYYY-MM-DD hh:mm:ss,m - octolapse.snapshot - ERROR - An unexpected exception occurred while post-processing images for the Webcam - Default OctoPi 0.16.0 camera.  See plugin.octolapse.log for details
Traceback (most recent call last):
  File "~/oprint/lib/python3.9/site-packages/octoprint_octolapse/snapshot.py", line 485, in create_thumbnail
    img.thumbnail([basewidth, hsize], Image.ANTIALIAS)
AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'
```
Next edit the file. Should be the line number shown above.

```
$ vi ~/oprint/lib/python3.9/site-packages/octoprint_octolapse/snapshot.py
```

That should get you there.

rickgode, 2023-08-20:
I tried the less ~/.octoprint/logs/plugin_octolapse.log but didn't have any errors.  So I guess mine isn't the same issue.  What I'm seeing is my timelapse preview is of something I printed before the last update.  It won't change.  Although the plugin seems to be working after the last update.  Thoughts?

Ricky

xXxH3LP3RxXx, 2023-08-20:
> I tried the less ~/.octoprint/logs/plugin_octolapse.log but didn't have any errors. So I guess mine isn't the same issue. What I'm seeing is my timelapse preview is of something I printed before the [...]

Can u upload the octolapse log?

Also i did this "fix" today & the preview is now working👍🏼

rickgode, 2023-08-22:
Mine doesn't seem to be the same issue, it just doesn't update the preview.  The image is from the last print before I updated octolapse to the latest version

I've uninstalled octolapse and will attempt reinstalling it.

[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/12404043/plugin_octolapse.log)
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/12404057/octoprint.log)

xXxH3LP3RxXx, 2023-08-22:
> Mine doesn't seem to be the same issue, it just doesn't update the preview. The image is from the last print before I updated octolapse to the latest version
>
> I've uninstalled octolapse and will [...]

Could be still the same issue, that log is pretty clean, did you log everything/errors? And start a print with it?

mattaster, 2023-08-22:
This seems like a different issue than #917, I'd suggest a new issue to keep things clean.

rickgode, 2023-08-31:
I can confirm now, that my issue was the same others on this thread and the renaming the Image.ANTIALIAS to Image.Resampling.LANCZOS did the trick, but mine was a tad different (and I am total n00b in Linux) Oh and vi is the DEVIL

I typed in the less ~/.octoprint/logs/plugin_octolapse.log and got a similar message as the above post at line 485, but my location of the snapshot.py was different so I had to type the stuff below to find it

vi /home/pi/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/snapshot.py

Then the arrow keys and such don't work the same in vi (because it's the DEVIL)

So I actually edited it in nano like this

nano /home/pi/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/snapshot.py

Thanks!

idaband, 2023-11-08:
This fixed the issue for me as well.  Will this be implemented in the next release by chance?

ninjamojo, 2023-11-08:
Going to try to get this added today.  I think there is a PR I need to merge.

idaband, 2023-11-09:
> Going to try to get this added today. I think there is a PR I need to merge.

awesome saw the change in the release this issue can be resolved.

FormerLurker, 2023-11-09:
Thanks for responding @idaband!  @CodedSakura  or anyone else, can you verify the issue is resolved after updating to the latest release?

CodedSakura, 2023-11-10:
can confirm, the issue is fixed in the latest release!

closed 2023-11-10 by CodedSakura

xamiell, 2024-02-29:
Hi,
Having this issue creating thumbnails

```
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_octolapse/snapshot.py", line 339, in process
    self.create_thumbnail()
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_octolapse/snapshot.py", line 484, in create_thumbnail
    raise SnapshotError(
octoprint_octolapse.snapshot.SnapshotError: snapshot-thumbnail-create-error: Create Thumbnail - An unexpected exception occurred while creating a snapshot thumbnail for the Webcam - Default OctoPi 0.16.0 camera.  Check the log file (plugin_octolapse.log) for details. - Inner Exception: module 'PIL.Image' has no attribute 'ANTIALIAS'
```

OctoPi new camera stack 1.0.0
OctoPrint 1.9.3
Python 3.9.2