FormerLurker/Octolapse #918: mk3s+ with mmu2s jamming on unload at end of print with smart compatibility

JoHnEyAp, 2023-07-17:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.43

#### Version of OctoPrint

OctoPrint Version: 1.91

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. setup octolapse correctly for prusa slicer v2.6 (verbose, profiles etc)
2. sliced a benchy with default settings, besides the changes required.
3. settings for stabalization are center left
4. settings for trigger is smart compatibility

#### What should have happened?
Benchy should have printed just fine, the artifacts may be a result of the smart compatibility choice.  The filament should have unloaded without jamming.

#### What happened instead?
On the smoke stack of benchy their is a long tail.  When the print finishes, the filament should unload without error and without ramming.

#### Operating System running OctoPrint and Octolapse

OS Name: Raspberry Pi 4b 8gb
Os Version: Raspberry Pi 4b 8gb

#### Printer model & used firmware incl. version

Printer Model: ___Prusa mk3s+ with MMU2S
Printer Firmware Version: mk3s - 3.12.2,   MMU2S - 3.12.2

#### Browser and version of browser, operating system running browser

Browser: google chrome
Browser OS: windows 10 PC

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

JoHnEyAp, 2023-07-17:
smart - high quality doesn't do this.  

only smart - compatibility from my testing.

Between the two profiles, their are the same number of snaps.  The only thing I can see that is different is high quality isn't causing the double retract which rams the tip at the end prior to unload.

Smart compatibility also does some weird things from what i noticed.  At the start, and sometimes throughout the print, it moves to take a photo then back to prime line then back to build.....i have no idea why

I do not clean the mmu or printer after this jam, re print with high quality or nothing and its fine.

is it possible prusaslicer 2.6 did something because of mk4?

I have tried g90/91 influence and that caused a firmware crash at the jam, weird letters and symbols appeared and wouldnt go away, it was definitely related to this setting....printer was doing weird things too

JoHnEyAp, 2023-08-06:
i formatted my sd and reinstalled octoprint and octolapse, everything seems to be working now.  I did not install the plugin that allows me to see the ip address on the mk3 this time, maybe that has a conflict? im not sure, previously my extruder was acting weird, now i notice it take the photo during infill, before it would take it before the print started.

I can provide more info if needed, but i think it is safe to say this was a corrupted install, or a conflict with another plugin

JoHnEyAp, 2023-09-06:
So the issue is still here after formatting the sd card.  I have found that it is related to the smart triggers, they do not work.

by using classic--->layer height--->only when lifted, this works just fine.

the smart triggers are causing horrible jams, the last one required disassembly of the hot end as a large piece got stuck.
it is definitely the smart triggers as I have 0 issues without octolapse and just octoprint or sd card, and now, so far, 0 issues with the above settings.

perhaps I'm using the smart triggers wrong, but even default is causing these issues.

I'm pretty sure its related to prusaslicer 2.6 or fw 3.12, i havent updated to 3.13 yet, but its beeping at me

I currently have automatic slicer configuration turned off, but i will enable it as i feel confident its working to my expectations

if i can provide any more info or help in anyway, please let me know, this is a great tool and i would love to see the bugs squashed

FormerLurker, 2023-11-08:
I think I saw your issue on Reddit and responded.  My MMU2 is not with me know, and i won't have access for a good long while (possibly late winter, but maybe not until spring).  Can you try an older version of prusa slicer?  I'm 99% sure it's related to the end gcode, so if there are some changes to the defaults there, it might make figuring this out easier.

JoHnEyAp, 2023-11-13:
Yes that’s me for sure.  I will try that and also think its the same
thing…..this made mk4 compatible along with tree supports….ill try it out
and let you know the results……would you prefer email or back to Reddit?

FormerLurker, 2023-11-16:
Here is better.  It's easier to track issues and code changes and stuff.

JoHnEyAp, 2023-12-29:
Hey, I haven't forgotten about this.

I had xmas orders and now family in town, I'll be back at it Jan 4th.

Before I start again, do you want me to change anything with your current
suggested setup?

A few new prusaslicer versions have released, I'll try an older one like u
said, but what about newer?

Just wanted to let you know why I haven't replied....I should have sent
this much earlier and I apologize for that

Thanks again for everything