FormerLurker/Octolapse #924: Octolapse won't install from Plugin Manager

Josh2ndBrain, 2023-10-14:
Hi, (I have reported an issue - first time here)

So I have been using Octolapse for about 15months now with no issue. The other night we had an extended blackout and my phone died during a print - I used my phone to run Octo4a v1.1.1. This has happened before, but this time when I went to login in to Octoprint, Octoprint kept thinking it was a brand new install by showing me the setup wizard - so I lost all my settings, saved prints ect, it wouldn't even recognise my OCtoprint login details... 

So I went through the wizard, re-set it up and then slowly started to install the few plugin I use - one of which is Octolapse... Every other plugin I wanted would install, however Octolapse would keep giving me an error! 

So I tried a couple of troubleshooting ideas... I installed Octo4a v1.2.5, re-set up Octoprint again, tried installing Octolapse plugin - ERROR. 

I factory reset my phone and then installed Octo4a v1.2.5, re-set up Octoprint again, tried installing Octolapse plugin - ERROR. 

I don't understand, because it was working fine for so long, but now all of a sudden it just won't worK????? Can someone review my error code below and see what I'm missing?

Thank you! 

[Octolapse error.txt](https://github.com/FormerLurker/Octolapse/files/12908155/Octolapse.error.txt)

coryroyce, 2023-10-24:
I'm having the same issue and would love to know if someone has a fix for this.

FormerLurker, 2023-11-08:
This should be fixed.  Please let me know and close if it starts working, thanks!

nguitard, 2023-11-25:
Hello I just tried to do a fresh install and it's still not working for me through plugin manager.

spanzetta, 2023-11-25:
same here.. it stays forever in this state:

1.4.0->Octolapse==0.4.5) (1.15.1)
Requirement already satisfied: pycparser in /home/pi/oprint/lib/python3.9/site-packages (from cffi>=1.0.1->argon2-cffi-bindings->argon2-cffi<22,>=21.3.0->OctoPrint>=1.4.0->Octolapse==0.4.5) (2.21)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /home/pi/oprint/lib/python3.9/site-packages (from packaging<24,>=21->limits>=2.8->Flask-Limiter<4,>=3.3.0->OctoPrint>=1.4.0->Octolapse==0.4.5) (3.0.9)
Building wheels for collected packages: Octolapse
Building wheel for Octolapse (setup.py): started

Josh2ndBrain, 2023-11-26:
Hi I tried it again and I still got an error

guardian606, 2023-11-26:
i have tried every thing to install here are my logs 

[octoprint.log](https://github.com/FormerLurker/Octolapse/files/13467469/octoprint.log)
[plugin_pluginmanager_console (1).log](https://github.com/FormerLurker/Octolapse/files/13467472/plugin_pluginmanager_console.1.log)

spanzetta, 2023-12-03:
still no news? Is it planned a fix?

FormerLurker, 2023-12-03:
@spanzetta, I just got the first indication that the last patch didn't solve everything 8 days ago.

The error posted by @nguitard is not the same as OP, and the one posted by @guardian606  is not the same as was posted by @nguitard.  It's a bit complicated to figure out in other words.

@nguitard:  The compile error in your logs is so strange.  Will try to replicate.

@guardian606: I think you might be missing some prerequisites.  Have you added the 'Plugin extras'?

If anyone with any c++ and python experience (preferably python extension experience) wants to try their hand at this I would be most appreciative.

FormerLurker, 2023-12-03:
@nguitard, it's going to take me a while to replicate, but I did take a stab at one possible thing:  added scope resolution to static call.  It's possible this is what your compiler is complaining about (even though it's a static call in the same class).  If so, there will be more (but hopefully different) errors.  Please keep feeding me setup py instructions.

FYI, you will need to install from the devel branch using the plugin manager.  You can use the *from url...* option and paste in this link:  https://github.com/FormerLurker/Octolapse/archive/refs/heads/devel.zip

guardian606, 2024-02-16:
@FormerLurker im having this issue on windows 10 not moble

guardian606, 2024-02-17:
ok i got it to install dnt know what fixed it but what i did was uninstall and reinstall python3 like 6 time and updated pip like 6 time then uninstalled and reinstall octoprint like 3 and it installed so i dnt know what fix it for me but it works now XD

spanzetta, 2024-02-19:
I also was then able to install the update..  without doing anything special..