FormerLurker/Octolapse #928: [Request] Allow compatability with Orcaslicer for automatic slicer configuration

yourbuddydinec, 2023-11-24:
#### If this is a feature request describe it here
I noticed when trying to use Orcaslicer with Octolapse it gives me the error:
Some slicer settings were missing from your gcode file. Missing Settings: No extruder settings were found in your gcode file.
even though I have verbose Gcode turned on. I could not find any other sections needed to turn on for this so hoping something can be added for this

#### Version of Octolapse
0.4.5

#### Version of OctoPrint
1.9.3
Used with Klipper

tcaruth, 2024-01-01:
I see [here](https://github.com/FormerLurker/Octolapse/blob/master/octoprint_octolapse/data/lib/c/gcode_comment_processor.cpp#L312) that Octolapse is doing pretty simple string searching. Looking at the gcode Orcaslicer puts out, I've gathered some of the types from a few of my prints:
```
;TYPE:Custom
;TYPE:Skirt
;TYPE:Support
;TYPE:Inner wall
;TYPE:Outer wall
;TYPE:Bottom surface
;TYPE:Internal solid infill
;TYPE:Sparse infill
;TYPE:Top surface
;TYPE:Overhang wall
;TYPE:Internal Bridge
;TYPE:Support interface
;TYPE:Bridge
```

Toastypaws, 2024-03-29:
I'm having the same problem and am wondering when this will be fixed and or added as a feature.

ishimarumakoto, 2024-04-26:
+2 here

Xantha123, 2024-06-28:
Is there any update on this? I have the same issue

Mukul1127, 2024-07-02:
> Is there any update on this? I have the same issue

I've had great success by just choosing custom and filling in the values

Villain-GH, 2024-08-02:
Simply enable "Verbose G-Code" under the "others" tab in Orcaslicer, near the bottom of the page.

dhelbegor, 2024-08-12:
Any soluting regardles to this problem? I'm having the same issue. even with the Verbose G-Code checked and saved. the error bellow appears every time.

"Some slicer settings were missing from your gcode file. Missing Settings: No extruder settings were found in your gcode file."

Fail-Safe, 2024-08-20:
> Simply enable "Verbose G-Code" under the "others" tab in Orcaslicer, near the bottom of the page.

This did not work for me. Could you elaborate on any other settings/options you set to get this working?

destaben, 2024-09-22:
same here, didn't work enabling Verbose G-Code

moeffju, 2024-11-04:
Unfortunately the automatic configuration detection is not very well documented, so I dug through the source code for a while and here is a solution for OrcaSlicer & Octolapse:

Octolapse is looking for the following settings: https://github.com/FormerLurker/Octolapse/blob/master/octoprint_octolapse/settings_preprocessor.py#L767-L778

`version` is automatically extracted from the header of the Gcode file, where it says

```
; generated by OrcaSlicer 2.1.1 on 2024-11-01 at 10:11:38
```

and this seems to work fine. The rest of the settings, Octolapse will look at the beginning and end of the gcode file for, in a format like `; setting_name = value`. This fails with OrcaSlicer because of several potential reasons: The thumbnail block is added to the beginning of the file, and can crowd out settings, and the config block at the end of the file is very large, so the settings Octolapse looks for might not be in the search window.

An easy way to fix it is to add the following code to the **Machine G-code** of your printer in the **Machine end G-Code** section:

```
; machine_extruder_count = 1
; --- Octolapse Settings
; retract_length = {retraction_length[current_extruder]}
; retract_lift = {retract_lift_above[current_extruder]}
; deretract_speed = {deretraction_speed[current_extruder]}
; retract_speed = {retraction_speed[current_extruder]}
; travel_speed = {travel_speed}
; first_layer_speed = {initial_layer_speed}
; layer_height = {layer_height}
; spiral_vase = {spiral_mode}
```

NB: This is only tested for single extruder machines as that is all I have access to. A similar approach should work for multiple extruders but you will need to add Jinja script to iterate.

Additionally, once you add this, you should also be able to disable Verbose G-code, if you want to make your files much smaller - but keep in mind that Octolapse uses the Verbose G-code comments for Smart triggers.

MarsBars06, 2025-02-01:
Hi all,

I've been stuck on this for ages but I can confirm that @moeffju 's method does indeed work.  Tested with orca slicer 2.3.0dev and the latest stable version of octolapse (0.4.5 as of writing).

That being said this is obviously a workaround and some updates to the documentation are needed with regards to usage with orca slicer, but it does work.

Once again, thanks to @moeffju for providing a working solution.  👍🏾