FormerLurker/Octolapse #932: Octolapse Install issue on Debian Bookworm

joebo69, 2024-01-09:
#### Version of Octolapse
Octolapse Version: v0.4.5

#### Version of OctoPrint
OctoPrint Version: v1.9.3

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: No, install issue

#### What were you doing when the problem occurred
1. Octoprint Settings, plugin manager, get more, octolapse, install
2. wait

#### What should have happened?
Octolapse should have installed without issues

#### What happened instead?
It did not install. Exited with error -
 
ERROR: Failed building wheel for Octolapse
Building wheel for Octolapse (setup.py): finished with status 'error'
Running setup.py clean for Octolapse
Failed to build Octolapse
ERROR: Could not build wheels for Octolapse, which is required to install pyproject.toml-based projects
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

#### Operating System running OctoPrint and Octolapse
OS Name: dietpi (debian Bookworm) on RPi 4 Model B 2GB (aarch64)
Os Version: v8.25.1
I suspect this may have something to do with either bookworm or the legacy vs. new camera stack. I am not using mjpg_streamer, I'm using a Raspberry Pi camera v3 which does not use the legacy stack.

#### Link to settings.json
Link to settings.json with all passwords removed: N/A

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  N/A

#### Link to octoprint.log
Link to octoprint.log:  Install log: <script src="https://gist.github.com/joebo69/9719da377ad44557ab7e0240e8cd2359.js"></script>

#### Screenshots and/or videos of the problem:
none required
Screenshot/Video Links: N/A

Larry2368, 2025-08-01:
I had the same issue.  I wound up installing v0.34.

rushiiMachine, 2025-08-02:
Installing v0.4.4 worked for me:

https://github.com/FormerLurker/Octolapse/archive/refs/tags/v0.4.4.zip

Larry2368, 2025-08-05:
> Installing v0.4.4 worked for me:
> 
> `https://github.com/FormerLurker/Octolapse/archive/refs/tags/v0.4.4.zip`

I found after I posted this 0.3.4 installed but I can not enable it.

Grr!