FormerLurker/Octolapse #934: Error downloading rendered videos

jcleek, 2024-01-15:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.5

#### Version of OctoPrint

OctoPrint Version: 1.9.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes (Log Everything)

#### What were you doing when the problem occurred

1. Print file
2. File Prints
3. Go to the VIDEOS AND IMAGES page for OctoLapse
4. Click the download link on a video

#### What should have happened?
The video should download

#### What happened instead?
Download icon changed to a red warning triangle. When I hover over that I see "An error occured while downloading your file. Click to retry. Error: 500 (Internal Server Error)"

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 1.0.0cam (build 2023.05.23.124648)

#### Printer model & used firmware incl. version

Printer Model: Prusa MK4
Printer Firmware Version: 5.1.2+13478

#### Browser and version of browser, operating system running browser

Browser: Brave, Chrome, Firefox, Edge, all updated with current versions.
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://getcis-my.sharepoint.com/:u:/p/jcleek/EcJfgEtjycJDruqchDZtS4QBlPteDodRhDx4JrT083vJ5w?e=dw6335

#### Link to settings.json

Link to settings.json with all passwords removed: https://getcis-my.sharepoint.com/:t:/p/jcleek/ESOIHfX6FltAiX5f3TWdS-sBPcFD-Yfdmb_r_sBlWZOMeg?e=tGv4jn

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://getcis-my.sharepoint.com/:u:/p/jcleek/ESPw9IczB75LmUnPmt_6--ABcEQUsI47yIx2gCTxobXPsA?e=bmAUsv

#### Link to octoprint.log

Link to octoprint.log:  https://getcis-my.sharepoint.com/:u:/p/jcleek/ESqvcfvvBNJGrTdWdPIxtG0BDdA_QB6bGrtLx_HzLsIVsQ?e=aEFA2j

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://getcis-my.sharepoint.com/:i:/p/jcleek/EZ8a90tH4ylDvCkLOZ7J7nUBvfaP6wD80OeO2v-x_WNEpw?e=NnUmGO

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

jcleek, 2024-01-15:
I continue to troubleshoot and using a second SD card I started over. A brand new install of Octoprint with only Octolapse installed and I get the exact same behavior. Using the new camera stack image from the Raspberry Pi imager I cannot get your plugin to render a timelapse.

jcleek, 2024-01-15:
I have been able to see a timelapse preview and it plays in the window, but I have no ability to download a file.
[image]

jcleek, 2024-01-15:
I took a photo of the message that comes up when I hover over the red warning. I have tried from multiple browsers on several Windows computers, all with the exact same behavior.
[image]

gatonero, 2024-02-28:
I guess it's the "&" in the file name. I have had similar problems with "#". After removing it no more trouble.