FormerLurker/Octolapse #937: [Request] Automatic Slicer Settings for Bambu Studio

minding-myown-business, 2024-01-20:
#### If this is a feature request describe it here

I would like the ability to run OctoLapse with files sliced in Bambu Studio. Whenever I try this, I always get this error:
[image]
I looked at the [wiki page regarding this](https://github.com/FormerLurker/Octolapse/wiki/Automatic-Slicer-Settings) but could not find any configuration for Bambu Studio, meaning the only real option would be using Manual Slicer Settings. I frequently switch between slicers, and could see this being a massive pain for me to switch profiles every time I print with Bambu Studio gcode or Cura gcode, or potentially ruining the day of a 3D printing noob with a BambuLab printer or just slicing with Bambu Studio without knowing they need to manually input slicer settings.

#### Version of Octolapse
 
Octolapse Version: OctoLapse 0.4.5 (latest at the time of writing this)

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.9.3 (latest at the time of writing this)

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

jneilliii, 2024-01-21:
Have you tried OrcaSlicer yet? That has the necessary option for Verbose G-Code output wanted by OctoLapse.

[image]

minding-myown-business, 2024-01-21:
> Have you tried OrcaSlicer yet? That has the necessary option for Verbose G-Code output wanted by OctoLapse.

I had not known that OrcaSlicer had the verbose G-Code feature, thanks!
I had looked into it a bit (but clearly not enough!) and saw that it was a fork of BambuLab (with the optimal algorithms, which was why I chose it), which was itself a fork of PrusaSlicer, which was a fork of Slic3r, at which point it becomes a bit of a nightmare for security/feature updates. 
I will look into it further though, and especially how it is on security and feature updates. 
Thanks for the suggestion!

Andyman14159, 2024-01-30:
I am currently trying to get it to work in Orca. I have verbose turned on and it still does not find the extruder settings... I am actively trying to find a cure for this.

Andyman14159, 2024-01-30:
I was able to get it to work with Orca by defining manually the extruder settings. Select other as your slicer in your print profile and then manually fill out the extruder details.

ishimarumakoto, 2024-04-26:
> I was able to get it to work with Orca by defining manually the extruder settings. Select other as your slicer in your print profile and then manually fill out the extruder details.

Can you post your settngs?

tcaruth, 2024-04-27:
#928 is also a request for Orcaslicer support