FormerLurker/Octolapse #942: Receiving Error when starting to Print - Some slicer settings were missing from your gcode file.   Missing Settings: Retraction Length,Retraction Speed,Deretraction Speed,Z Lift Height,X/Y Travel Speed,Z Travel Speed

bretthatcher, 2024-03-08:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

rgaborko, 2024-05-05:
My Octolapse was working perfectly fine until last week and now it is coming up with the same error.  gcode was copied and pasted and was working great now it is not and had to disable it to be able to print.

MarsBars06, 2024-06-22:
@bretthatcher 
I think it may be because Octolapse searches the first and last 1000 lines of code.
In my case, I use a thumbnail plugin in Cura and it adds lines into the generated gcode file **before** the start gcode that is specified in the machine settings section in Cura.
I have yet to verify this and I will get back to you, but its probably worth opening one of your gcode files that triggers said error and open it in notepad++ (or similar) and just check what lines the octolapse comments are on.