FormerLurker/Octolapse #947: Octolapse Startup Failed - Automatic Configuration - Ender 3 V2 Neo

atkeeler, 2024-04-29:
I've been trying to get Octolapse set up with my Ender 3 V2 Neo for a while and I can't seem to get it working, continuously getting the message provided in the attached screenshot. The slicer I'm using is Orca Slicer, as I've found it works better with my V2 Neo running MRiscoC 2.1.3 custom firmware. I also went ahead and attached the plugin log file for additional information.
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/15154469/plugin_octolapse.log)

[image]

atkeeler, 2024-04-29:
I wanted to include my start and end G-code for a reference in case anyone asked.

Start G-code:
`M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
M140 S[bed_temperature_initial_layer_single] ;Set final bed temp
M104 S[nozzle_temperature_initial_layer] ;Set final nozzle temp

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
M190 S[bed_temperature_initial_layer_single] ;Wait for bed temp to stabilize
M109 S[nozzle_temperature_initial_layer] ;Wait for nozzle temp to stabilize
G1 X10.1 Y145.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y145.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0  ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0.0000 F1800`

End G-code:
`G91 ;Relative positionning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

G1 X0 Y0 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z`

dhelbegor, 2024-08-12:
Dang, asked on apr and no response yet :(.

I'm about to give up. i'm interested in using orca slicer too. and I'm having the same error. they say to activate the verbose G-code in orca slicer but that didin't work at all.

@atkeeler did u find a solution ?