FormerLurker/Octolapse #949: [Bug]: Cannot install on Python 3.12

JohnyPeaN, 2024-06-11:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.45

#### Version of OctoPrint

OctoPrint Version: 1.10.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Upgraded to octoprint 1.10.1, system upgraded python to 3.12
2. Tried to install octolapse plugin - failed
3. Prepared latest versioneer.py from master (https://github.com/python-versioneer/python-versioneer)
4. Extracted octolapse.zip, replaced versioneer.py with vesion from step 3, repacked into zip
5. installed from resulting file - success

#### What should have happened?
Installation should succeed.

#### What happened instead?

> Collecting https://github.com/FormerLurker/Octolapse/archive/v0.4.5.zip
Downloading https://github.com/FormerLurker/Octolapse/archive/v0.4.5.zip
/ 1.3 MB 4.8 MB/s 0:00:00
Preparing metadata (setup.py): started
error: subprocess-exited-with-error

>× python setup.py egg_info did not run successfully.
│ exit code: 1
╰─> [15 lines of output]
/tmp/pip-req-build-qf456kab/versioneer.py:421: SyntaxWarning: invalid escape sequence '\s'
LONG_VERSION_PY['git'] = '''
Traceback (most recent call last):
File "<string>", line 2, in <module>
File "<pip-setuptools-caller>", line 34, in <module>
File "/tmp/pip-req-build-qf456kab/setup.py", line 25, in <module>
plugin_version = NumberedVersion.clean_version(versioneer.get_versions(verbose=True)["version"])
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/tmp/pip-req-build-qf456kab/versioneer.py", line 1412, in get_versions
cfg = get_config_from_root(root)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/tmp/pip-req-build-qf456kab/versioneer.py", line 342, in get_config_from_root
parser = configparser.SafeConfigParser()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
[end of output]

>note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

>× Encountered error while generating package metadata.
╰─> See above for output.

>note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Preparing metadata (setup.py): finished with status 'error'`

#### Operating System running OctoPrint and Octolapse

OS Name: Manjaro ARM Linux
Os Version: 24.04

#### Printer model & used firmware incl. version

Printer Model: Anycubic i3 Mega S
Printer Firmware Version: Marlin 2.1.2.2 (knutwurst 1.5.4)

#### Browser and version of browser, operating system running browser

Browser: Firefox
Browser OS: Cachyos

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  not available / not installed

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

uranoxyd, 2024-06-24:
Same here (Python 3.12.4)

Firenyth, 2024-08-05:
I have the same issue, how did you prepare the versioneer.py file?

mrkylegp, 2024-08-07:
Same issue here.

Big-Channie, 2024-10-26:
Same issue here as well

UberEclectic, 2025-04-24:
Same issue here.

jneilliii, 2025-07-16:
If my assumption is correct about attaching master.zip to releases, then the software update hook will need to be tweaked as well to handle this new requirement. Not sure how this will impact your pre-release checks, etc. but I think it should still work. 

```py
    octolapse_update_info = dict(
        displayName="Octolapse",
        # version check: github repository
        type="github_release",
        user="FormerLurker",
        repo="Octolapse",
        pip="https://github.com/FormerLurker/Octolapse/releases/download/{target_version}/master.zip",
        stable_branch=dict(branch="master", commitish=["master"], name="Stable"),
        release_compare='custom',
        prerelease_branches=[
            dict(
                branch="rc/maintenance",
                commitish=["master", "rc/maintenance"],  # maintenance RCs
                name="Maintenance RCs"
            ),
            dict(
                branch="rc/devel",
                commitish=["master", "rc/maintenance", "rc/devel"],  # devel & maintenance RCs
                name="Devel RCs"
            )
        ],
    )
```

jneilliii, 2025-07-17:
Ok, I take that back. I think the issue stems from not having a tagged version on the master branch after merging the changes in from the PR. Without the tag on the master branch versioneer has no way of knowing what's going on. For anyone needing to install I created a v0.4.5 tag on my fork that can be installed with the following URL. Copy/Paste into Plugin Manager > get more > ...from URL

```
https://github.com/jneilliii/Octolapse/archive/refs/heads/master.zip
```