FormerLurker/Octolapse #957: Unable to install on a clean Ubuntu System

XeroZar, 2024-10-28:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___0.4.5 (latest available)

#### Version of OctoPrint

OctoPrint Version: ___0.4.5

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. __Trying to install on a clean Ubuntu System
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___Should have installed like the other plugins I installed.

#### What happened instead?
___Got the following error when installing...
Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/master.zip...
/home/pi/OctoPrint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpzb_vxm70/Octolapse-master.zip --no-cache-dir
Processing /tmp/tmpzb_vxm70/Octolapse-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'error'
error: subprocess-exited-with-error

× python setup.py egg_info did not run successfully.
│ exit code: 1
╰─> [6 lines of output]
Traceback (most recent call last):
File "<string>", line 2, in <module>
File "<pip-setuptools-caller>", line 34, in <module>
File "/tmp/pip-req-build-ynvf7rgv/setup.py", line 6, in <module>
from distutils.msvccompiler import MSVCCompiler
ModuleNotFoundError: No module named 'distutils.msvccompiler'
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

#### Operating System running OctoPrint and Octolapse

OS Name: ___Ubuntu
Os Version: ___24.04.1

#### Printer model & used firmware incl. version

Printer Model: ___Ender 3 Pro
Printer Firmware Version: ___2.1.1

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

rolandbough, 2024-11-02:
I have this same issue fresh install of same version of Ubuntu.

Big-Channie, 2024-11-05:
I have this same issue. I believe it's a python version problem. Hoping this gets fixed soon

rolandbough, 2024-11-10:
> I have this same issue fresh install of same version of Ubuntu.

Looking forther into it, it seems with Python 3.12 (default for Ubuntu 24 Server) the distutils in general is depreciated. So there is no package for it, this has been totally replaced by setuptools in python 3.12.

As a result when pip tried to install the requirement for distutils.msvccompiler it doesn't exist, there should be something similar under setuptools.