FormerLurker/Octolapse #966: Can't install Octolapse from repository

romanholtwick, 2025-07-02:
I have a fresh install of the newest stable Octopi (1.1.0) on a Raspberry Pi 3B, tried to install Octolapse from the repository, but it always cancels.
I tried a number of things, but I can't get it to work.
Here's the log:

`Installiere Plugin "Octolapse" von https://github.com/FormerLurker/Octolapse/archive/master.zip...
/opt/octopi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmp9rhf4s6p/Octolapse-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmp9rhf4s6p/Octolapse-master.zip
Preparing metadata (setup.py): started
error: subprocess-exited-with-error

× python setup.py egg_info did not run successfully.
│ exit code: 1
╰─> [31 lines of output]
/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/dist.py:548: UserWarning: The version specified ('refs/pull/718/head') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
warnings.warn(
Found packages: {'octoprint_octolapse', 'octoprint_octolapse_setuptools', 'octoprint_octolapse.test'}
running egg_info
Traceback (most recent call last):
File "<string>", line 2, in <module>
File "<pip-setuptools-caller>", line 34, in <module>
File "/tmp/pip-req-build-lnaoa20n/setup.py", line 130, in <module>
setup(**setup_parameters)
File "/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/__init__.py", line 108, in setup
return distutils.core.setup(**attrs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
return run_commands(dist)
^^^^^^^^^^^^^^^^^^
File "/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
dist.run_commands()
File "/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
self.run_command(cmd)
File "/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/dist.py", line 1213, in run_command
super().run_command(command)
File "/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 987, in run_command
cmd_obj.ensure_finalized()
File "/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 111, in ensure_finalized
self.finalize_options()
File "/opt/octopi/oprint/lib/python3.11/site-packages/setuptools/command/egg_info.py", line 219, in finalize_options
parsed_version = parse_version(self.egg_version)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/opt/octopi/oprint/lib/python3.11/site-packages/pkg_resources/_vendor/packaging/version.py", line 266, in __init__
raise InvalidVersion(f"Invalid version: '{version}'")
pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: 'refs-pull-718-head'
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Preparing metadata (setup.py): finished with status 'error'
Fehler!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`

FormerLurker, 2025-07-02:
> File "", line 2, in
> File "", line 34, in

See if you can install from this URL in the plugin manager (the ...From Url
box):
https://github.com/FormerLurker/Octolapse/archive/refs/heads/master.zip

sSpeaker, 2025-07-07:
I had the same error. [Release](https://github.com/FormerLurker/Octolapse/archive/refs/tags/v0.4.5.zip) worked for me.

evindunn, 2025-09-19:
Worked for me from the most recently tagged release: https://github.com/FormerLurker/Octolapse/archive/refs/tags/v0.4.5.zip